prusa3d/PrusaSlicer #3984: Layer numbers wrong in preview if travel moves is on

TobbeJ, 2020-04-04:
### Version
2.2.0+win4

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
Prusa I3 MK3S (v3.8.1) + MMU2S (v1.0.6)

### Behavior
Take any model and slice it, go to the preview and note the layer number at for example the top.
then click "Travel" checkbox to show travel moves.
Notice that now the layer number at the top is different and if you scroll all the way down to the build plate you now have a layer zero that was not there before plus you got a layer there in the beginning (2nd or 3rd depending on how you count) that looks empty.

#### Project File (.3MF) where problem occurs
Not needed, problem can be reproduced with just Add shape and for example a box

neophyl, 2020-04-04:
You have asked it to show travel moves.  These can happen on layers above your model/printable layers.  For example if your settings are to raise 0.6mm on z hop then on the last printable layer the nozzle would travel up 0.6mm (three layers on a 0.2 layer print) so it would have to display an extra 3 layers.  This isnt a bug, its a necessary part of what you have asked the preview to display.

The blank layer at the bottom *might* be a bug but I suspect its adding it to account for the travel of the nozzle to get from wherever its starting to the first layer for printing.

rtyr, 2021-02-09:
Should be fixed in 2.3.0. Closing.

closed 2021-02-09 by rtyr