prusa3d/PrusaSlicer #3985: Problem printing stacked objects

smurfix, 2020-04-04:
### Version
2.2.0

### 3D printer brand / version + firmware version (if known)
any with MMU

### Behavior

I want to print multiple mostly-flat objects on top of each other. The idea is to print one object, let the slicer add a layer of PVA, then the next object.

PrusaSlicer is unable to do this. It insists on adding support structures on the sides of the lower objects. This wastes a lot of filament and time, and frankly I wonder why it does this.

The attached 3MF has two same-sized cubes hovering directly on top of each other. Note how the lower cube is completely obscured by superfluous support structures.

[image]

It'd be nice if solving this could be expedited. I'm part of a team which prints parts for face shields and other items required for the effort to contain the COVID-19 pandemic. Fixing this problem would allow us to print a stack of parts through the night, instead of the printer sitting idle.

#### Project File (.3MF) where problem occurs
[Stacked Cubes test.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4432278/Stacked.Cubes.test.3mf.zip)

rtyr, 2020-04-04:
I am not sure if this is ideal approach. Here are some alternatives:
- turn off the supports completely, stack it without empty space between the parts, then use height range layer modifier or regular modifier and assign PVA extruder to it
- if you want bigger space between the parts, use another object to fill this space and asign PVA extruder to it (supports disabled)
- use alternative methods without any supports or PVA, for example https://www.prusaprinters.org/prints/27267-prusa-face-shield-stack-ver

You can also play with the pattern spacing parameter. See the picture below with pattern spacing set to 0.

[image]

Secondary problem you mentioned is not problem at all. That is the final ramming before the filament is pulled from the nozzle and unloaded. It is FAQ. https://github.com/prusa3d/PrusaSlicer/issues/1355

smurfix, 2020-04-05:
Your proposed solutions do not work because the slicer will assume a continuous print and not emit top and bottom infills at the PVA borders. That's not really an issue for most of the face shield models, but that's not the only thing I want to print that way.

Imperfect bonding between PETG and PVA can be fixed quite easily in the model, by adding a few strategically placed joiners.

I didn't have any luck with alternate "stacked" models, most likely because I'm using a 0.8mm nozzle and the supposedly-separate parts bind much too tightly to each other. I want to separate the parts, not rip them apart …

rtyr, 2020-04-05:
That is not a problem, you can force solid infill anywhere you want using the same technique.

What about the pattern spacing? Did it help?

[image]

smurfix, 2020-04-05:
Sure I can force 100% infill, but that doesn't help when the top infill should have a different pattern. Also, quite frankly, adjusting a model's height, changing the repeat count, and/or importing to the slicer takes 30 seconds. Having to play around in PrusaSlicer to move 30 infill modifiers to whichever layer they should be at is no fun, eats a lot of time, and is prone to errors.

Apparently it's also much too easy to crash the slicer when you do that. I'll field a separate report about that tomorrow.

Also I need an externally-scaled-down model for the PVA filler because, surprise, positioning a scaled-flat object requires actual math to get right:
[image]
The thing is supposed to be at zero mm height. Not float in the middle …

rtyr, 2020-04-05:
> 
> The bottom of any object is supposed to be at z=zero. Except when it's not?? In any case I'll file a bug report for that too.

 https://github.com/prusa3d/PrusaSlicer/issues/3579

bubnikv, 2021-06-03:
> Your proposed solutions do not work because the slicer will assume a continuous print and not emit top and bottom infills at the PVA borders. That's not really an issue for most of the face shield models, but that's not the only thing I want to print that way.

Enable mutliple objects -> Interface shells to get the solid infill between layers printed with different extruder.

Also PrusaSlicer 2.4.0-alpha1 will implement snug supports, thus your issue is a duplicate of 
    Fixes Support problems with models with hole in the walls. #555
    Fixes Support in the Air #740
    Fixes [Bug] Supports generated beyond bed edges (X<0 and X>250) and where none are needed. #902
    Fixes Unable to remove support material/can't change support "inflation distance" #2708
    Fixes FR: support inflation and support conform to boundary #4783
    Fixes Support blocker not working on this model #1346
    Fixes Unnecessary support material #1993

closed 2021-06-03 by bubnikv