prusa3d/PrusaSlicer #3986: Selecting all Version 2.2.0 configuration filaments in setup but none shown in slicer

m3cinc, 2020-04-04:
### Version
2.2.0
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Win64

### 3D printer brand / version + firmware version (if known)
MK3S+MMU2S

### Behavior
During install configuration, select all filaments, then finish configuration
When the slicer opens, go and select a filament... none is displayed

This is an apparent bug

rtyr, 2020-04-04:
I am unable to reproduce it on 2.2.0+win64. Are you sure, that you have correct printer profile selected on the main screen? Can you share some screenshots? What was your previous PrusaSlicer version?

doublec4, 2020-04-05:
I am having the same issue. Just downloaded Prusa Slicer 2.2 for the first time, no previous versions. I didn't select all filament profiles, only a few... but after finishing the config wizard I have no filament profiles. I have a custom printer set up, not a PRUSA profile.

rtyr, 2020-04-05:
> .... after finishing the config wizard I have no filament profiles. I have a custom printer set up, not a PRUSA profile.

Intended behavior. See https://github.com/prusa3d/PrusaSlicer/issues/3925

sberesfo, 2020-04-05:
I am having the exact same issue. My modified filament files imported fine from my earlier version but none of the standard filament profile loaded or will load.

rtyr, 2020-04-05:
> 
> 
> I am having the exact same issue. My modified filament files imported fine from my earlier version but none of the standard filament profile loaded or will load.

What printer profile are you using?

crokatron, 2020-04-12:
I'm having the same issue with an Ender 3, cant get any filaments to show up minus the Generic and Prusament ABS PETG and PLA

rtyr, 2020-04-12:
> 
> I'm having the same issue with an Ender 3, cant get any filaments to show up minus the Generic and Prusament ABS PETG and PLA

Intended behavior. Other filaments belong to different vendor space.

crokatron, 2020-04-12:
> > I'm having the same issue with an Ender 3, cant get any filaments to show up minus the Generic and Prusament ABS PETG and PLA
> 
> Intended behavior. Other filaments belong to different vendor space.

Understood - but where can I find these profiles so I can load and tweak them manually? I want to switch from S3D but to manually generate, from scratch, all of my profiles, again, will be my barrier to entry.

rtyr, 2020-04-12:
> 
> Understood - but where can I find these profiles so I can load and tweak them manually? I want to switch from S3D but to manually generate, from scratch, all of my profiles, again, will be my barrier to entry.

It is described in this comment https://github.com/prusa3d/PrusaSlicer/issues/3934#issuecomment-605428395. You might also need to delete "compatible printers condition" field.

neophyl, 2020-04-12:
To get you and others in a similar position stated with something to work from I put together the following  https://forum.prusaprinters.org/wp-content/uploads/2020/04/GenericFilaments_Neo_20200401.zip

Just unzip and File>import >import config bundle in Prusa Slicer.  That should add copies of all the filaments that have been detached from the vendor space and otherwise mostly freed from required by rules

rtyr, 2021-07-12:
No response. Other issues mentioned here are related to different vendor spaces. Closing.

closed 2021-07-12 by rtyr