prusa3d/PrusaSlicer #3989: COVID 19 Mask STL error

numberformat, 2020-04-05:
Trying to print a mask from 

https://www.thingiverse.com/thing:4225667

However Prusa slicer version 2.2.0-win64 says that the latest "Filter_standard_cap.stl" contains errors. Repairing the stl did not work either.

Empty Layers detected. the output would not be printable

Object name: Filter_standard_cap.stl
printz: 1.350000

This is usually caused by negligibly small extrusionn or by a fault model. Try to repair the model or change its orientation on the bed.

### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

numberformat, 2020-04-05:
They say STLs work fine in Cura and simplify 3d

rtyr, 2020-04-05:
Please provide your 3MF project file. Filter_standard_cap.stl works fine for me.

guestisp, 2020-04-05:
just use any stacked mask like this: https://www.thingiverse.com/thing:4244884/

numberformat, 2020-04-07:
Actually un-installed and re-installed Prusa Slicer and slicing the same file and its working fine now.

closed 2020-04-07 by numberformat