prusa3d/PrusaSlicer #3994: Extreamly large print times

PsyonicOne, 2020-04-05:
### Version
2.2.0+linux-x64

### Operating system type + version
Linux kernel-core-5.5.5-200
Fedora 31

### 3D printer brand / version + firmware version (if known)
Ender-3 mechanics (modified)
Marlin 1.1.8 firmware (modified)

### Behavior
The times shown after slicing are enormous! See pic. [times](https://user-images.githubusercontent.com/46242800/78472749-abcaca00-777e-11ea-976a-9a3441b4c59c.png)
All other details look right.
This did also happen in the 2.2.0 beta Linux version but it would correct itself after a few settings changes and re-slicing, so far I have not been able to make this correct in 2.2.0-stable-relese.

* _Steps needed to reproduce the problem_
Slice any STL

  * _If this is a command-line slicing issue, include the options used_

* _Expected Results_
Not stupid-crazy times

* _Actual Results_
See above pic

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[badTimes.zip](https://github.com/prusa3d/PrusaSlicer/files/4433450/badTimes.zip)

rtyr, 2020-04-05:
Seems OK on 2.2.0+win64

[image]

PsyonicOne, 2020-04-05:
Thanks mate. Someone else also verified that. Perhaps just a Linux issue.... Or, perhaps just something wrong with my install....
Oh, I'll update the report to include "Using .AppImage" version

rtyr, 2020-05-21:
I was able to reproduce it on Ubuntu 20.04 LTS, but only when using .AppImage version.

YMS137, 2020-05-22:
I had the same thing on Mac with Catalina
[image]

lukasmatena, 2024-02-18:
Cannot reproduce in 2.7.2-alpha2, and I remember a similar issue being fixed in the meantime. Closing.
We can reopen it if anyone can reproduce it.

closed 2024-02-18 by lukasmatena