prusa3d/PrusaSlicer #3997: [Feature request] Preview line progress

arbitur, 2020-04-05:
I wish to be able to see the progress of the lines being printed like in Cura where you can scrub a specific layer and see the order of every line segment in a sequence from start of layer to end of layer.
As of today in latest (2.2.0+win64) PrusaSlicer one can only see the entire layer at once.

Patronics, 2020-04-07:
One potential workaround until this is implemented, if you use OctoPrint for your printer, you can use the GCodeViewer in OctoPrint to see how it will move throughout the layer. There's probably other 3rd party software that can give you a similar preview to the gcode output by PrusaSlicer.

foreachthing, 2020-04-08:
I'm still using Craftware, if it becomes critical.
It shows the corresponding gcode as well:
[image]

bubnikv, 2020-10-17:
IMplemented in PrusaSlicer 2.3.0-alpha1.

closed 2020-10-17 by bubnikv