prusa3d/PrusaSlicer #4000: The infill for grid crosses itself on the same layer.

Hackpig1974, 2020-04-05:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 Professional

### 3D printer brand / version + firmware version (if known)
MK3

### Behavior
Using the default Prusa Print Settings for SPEED, uses Grid as infill. 
The infill for grid is horrible and crosses itself on the same layer. On the same layer it will print one direction, and then on the same layer go against that printed line. This is bad because it causes the nozzle to have to try and melt through the previous lines that were laid down. This causes the intersections of the pattern to become raised and the nozzle to bump as it goes over these lines. This sounds like a cheese grater and if using PETG, the cross section gets taller and taller over those cross sections. You can physically see the nozzle bounce as it crosses over. The printer should never print over an existing line at the same layer or height. 

I hope this makes sense. I have to create a custom profile for all of the SPEED settings that utilize Grid infill as I can't modify the Prusa defaults for those Print settings.

The Print Settings that are defaulted to Grid are: 
**0.15mm SPEED @MK3
0.20mm SPEED @MK3**

This infill method should be remove or changed to print one direction on 1 layer, and the opposite direction on the 2nd layer.

#### Project File (.3MF) where problem occurs
Any STL file that utilizes this infill type.

Hackpig1974, 2020-04-05:
Grid Infill. 2nd Pass causes it to jump the previous line. Each crossing of a line causes it to get taller and taller.
[image]

Rectilinear never crosses a line and prints perfect. Why is this not the default?

[image]

Every time Grid crosses an existing line on the same layer it causes stress on the motors and the belt as it encounters resistance. Grid goes against the logic of 3D printing as slicers are designed to not cross an already printed line on the same layer.

wavexx, 2020-04-05:
The same is done in Cura.
You will also notice that parts printed in rectilinear are not as strong as grid, especially on the Z axis.

Sadly, you're also right. Grid only works well at medium-low speeds. As speed is increased, the pressure-drop caused by the ridge creates a gap on the other end of the crossing, which will cumulate over multiple layers until it makes the infill completely useless.

Again, this is _especially_ noticeable at high fill rates, and I have a report or two for Cura itself about this. There are work-arounds. Using larger extrusions for infill for pieces that need to be printed fast works. The viscosity of the material matters: PETG fares better at this. One solution I proposed for Cura is to ensure that the grid is filled in alternate directions at each layer, so that crossings do not result in Z gaps. This is _hard_ to do.

Like most things in 3d printing, it's a trade-off you have to make. Grid is still a very good infill type for most prints IMHO

wavexx, 2020-04-05:
https://github.com/Ultimaker/CuraEngine/issues/697

Hackpig1974, 2020-04-05:
I totally agree with everything you are saying. My problem is that this is set as the default for speed. This particular infill pattern is actually the worst to use for speed. If they set to be used for Draft, that would make sense. Even though rectilinear only goes every other layer for the pattern, it takes longer. That makes no sense to me and blows my mind. Grid in theory should take longer as it does both swipes on each layer. But slicer doesn't show that when I changed between rectilinear and grid. Gird is faster. I really like using the Prusa defaults as I feel they have been tested and used in house. But this Grid infill has been bad for like the last 4 versions. Everytime I use with PETG and their default settings it sounds horrible and causes the nozzel to jump. I may just have to start using the Quality Print Setting Profile, but I am sure it comes at a sacrifice for other settings that effect speed. Grid may be stronger, but if it is printing against other lines on the same layer, it is not good practice and will long term cause unnecessary wear on the printer.

wavexx, 2020-04-05:
I know the problem, but I'm not so sure about changing grid anyway.
"Rectilinear" for me is for "drafts", at best. The pieces are often too weak to be useful.

I still use the default grid for speed for most of my prints. The important point is that it's actually fast also in complex geoetry and works - I don't say perfectly - decently in a good number of scenarios.

The "ragging" on the print is often caused by excessive volumetric speed IMHO. Even when using PLA, you'll notice that extruding at the default max setting of  15mm3/s the layer is not well melted. I think this is way too high for the V6. Using 12 or 11mm3/s limits the infill speed for thicker layers, but you'll notice that the almost disappears.

"Cubic" is a nice contender for a "speed" profile. It's still relatively fast, but the reduced number of crossings (and moving positions of each) reduce this effect quite a bit.

Hackpig1974, 2020-04-05:
I will test Cubic for my custom profiles I had to create. If Cubic works, they should make that the default for speed. No profile should cause the extruder to cross a line that has already been printed on the same layer. That's just bad business. LOL

wavexx, 2020-04-05:
It has it's role. Try reducing max volumetric speed in the filament settings -> advanced. If you want to keep the speed, and you're printing a piece that has a lot long infill, raising temperature will counter-balance this. It's a hardware problem as much as a software one.

Hackpig1974, 2020-04-05:
I hear what you are saying, but default profiles are supposed to be just that. The point of default is you are supposed to be able to grab them and print with little to no changes necessary. For PETG the SPEED profile ends up becoming a disaster if I don't slow it down even more for GRID. The profile is also labeled SPEED, not QUALITY or STRENGTH. I just want to be able to use the defaults with having to make modifications to make it work. Sounds like a different infill should have been selected for the default of SPEED.

Hackpig1974, 2020-04-05:
Checked all the profiles and for SPEED it seems that **CONCENTRIC** is the fastest in regards to print times. I am going to print some parts with it, as well as Cubic, and see how the print results are. Thanks for some of your inputs and I hope they make a change to the SPEED profiles.

wavexx, 2020-04-05:
It's not great I agree, but now I'm curious. I know the grinding on the print can happen, but I was printing with default for a relatively long time on the MK3 using some brands of PETG and it was relatively rare for me this have this. I had worse problems to worry about :)

Could you share with us a model that gave you this problem, and which filament brand you're using?

Hackpig1974, 2020-04-05:
Here is an example: https://www.thingiverse.com/thing:3528283

[image]

After enough time those buildups start to sound horrible and cause the nozzle to bounce as it goes over each intersection. It is noticeably worse with PETG. PLA the nozzle can just burn through each crossing a little easier. It bounces less, but you can hear it is it crosses each section. It's not super hard to reproduce.

wavexx, 2020-04-05:
Filament brands? This might be helpful for testing

Hackpig1974, 2020-04-05:
I print with the Prusament and Hatchbox mostly.

Photogad, 2020-04-15:
What infill pattern is best if you care most about quality and speed, but don't care about strength?  

I use gyroid for PLA and grid for PETG. I find PLA prints most infill good, but PETG sometimes struggles. I definetly get that raised grinding with grid and PETG, but sometimes I do with rectilinear too. 

Maybe my infill speeds to fast for PETG. Im at 55mms for PETG infill with my custom n profile, still slower than prusa default

wavexx, 2020-04-15:
If you're printing a decorative thing, I think gyroid is currently the best. The new perimeter connection in 2.2 does less overextrusion in tight corners compared to rectilinear/grid/cubic, which is something might bleed on the corners. It's a bit slower for the same % infill, but again - if that's looks we're talking about I think you can just reduce the infill get it up to the same speed as grid.

Photogad, 2020-04-15:
@wavexx  another problem is where it needs to print small solid layers on top of infill. The solid layers are so small they don't connect to the infill and just float over the gaps in the infill, a lot of times it curls up. Only solution I've found is to increase infill to 60% or higher.  

Better slicing is needed. This model did not really need solid layers around these holes. Idk why it put solid layers there

[image]

[image]

Hackpig1974, 2020-04-15:
This is why I quit using GRID infill.

Photogad, 2020-04-15:
> This is why I quit using GRID infill.

What do you use now instead? Maybe cubic is best option

Hackpig1974, 2020-04-15:
I have been using Concentric and Cubic for PETG. The speed is comparable most times with Concentric being faster usually. Those 2 infills don't cross printed lines on the same layer. This has worked well for me. I wish they would change the default Prusa profiles to not use GRID as a default for the Speed Profiles.

jspring88, 2020-07-09:
I found this thread while trying to find an infill type in Cure that doesn't cross over itself in the same layer.  I wasn't able to find it in a quick web search so I did some slicing myself and found the following:
Infill types that don't cross over themselves on the same layer:  Lines, Concentric, Zig-Zag, Cross, Cross 3D and Gyroid.
Infill types that cross over themselves on the same layer: Grid, Trisngles, Tri-Hexagon, Cubic, Cubic Subdivision, Octet and Quarter Cubic
I agree with wavexx. What it looks like to me is that when the nozzle hits the previously laid down line of infill as it crosses it, it wipes off the nozzle and then on the other side there is a void until the filament starts extruding again.

jspring88, 2020-07-09:
One thing you can try is "Gradual Infill Steps" under Infill in Cura's settings.  By putting a number in that box the infill is sparse low in the print but it increases as it gets close to the top of the area being infilled.

wavexx, 2020-09-01:
Had to print a large empty part yesterday, where for the sake of speed I've been using grid. This is some gray petg printed at ~100mm/s. I've taken the pic on the moving platform, so the image is not great:

[image]

You can actually spot the direction of the infill due to the shape of the crossings (notice the widening before the crossing, followed by thinning just after it). However, notice it's still perfectly connected and nicely fused (I wouldn't use it otherwise) and stays like this without degrading further. The important factor here is not exceeding the volumetric capacity of the v6. I'm using 8mm3/s for this material, with a .4 nozzle and .2 layers this is what I can do:

[image]

I would still run into the issue if I printed faster (in which case I would switch to something else) , but this is just to say that "grid" is not "worthless" and it's still a very useful and fast infill. It could be better, sure.

DjDemonD, 2020-10-01:
I would like to propose that selecting grid infill, with greater than 90% infill should throw an error code/popup warning. I'm printing a part now with 97% infill using grid and its grossly over-filling the infilled area, such that to complete the print without head/nozzle damage I'm having to reduce the overall extrusion multiplier on the fly to 90%, which Is weakening the perimeters.

closed 2026-09-29 by prusaslicer-bot[bot]