prusa3d/PrusaSlicer #4001: Feature request: random function for gcode macro language

wavexx, 2020-04-05:
I'd love to have a simple no-frills implementation of rnd() for the gcode macro language, returning a random float in range 0-1. Or rnd(max) if you prefer.

I'd like to randomize the position of the purge line a bit to avoid scarring the PEI too much on the same spot. Having such a function would allow me to choose a different spot and/or direction (from left to right or vice-versa) without any additional feature.

_Is this a new feature request?_
Yes

gitcnd, 2020-10-31:
LOL - I just had the same thought - I've not tested it yet, but I did this :-

#555 = #555 + 1 ;This is a permanent variable (saved between power cycles) - add one, so our prime doesn't over-write the last prime
#555 = #555 MOD 25 ; keeps our parameter within the range 0..24
#556 = #555 + 15 ; add 15mm so I do not prime on the very edge
G1 X10 Y#556 F1500 ; move to prime
... prime code here...

I'm assuming Y#556 will tell my printer to move the Y axis to the value of the variable #556 - and I'm assuming my FFF gocde supports both permanent variables and the MOD function...

ardenpm, 2020-11-22:
I've just added this and will submit a pull request soon. It will add a macro function `random(min,max)` which can be used to give a value in the given range, double and integer variants.

gitcnd, 2020-11-22:
AWESOME!!  My gcode idea using permanent variables did not work out... the variables turned out *not* to be permanent - I guess they simply never supported that when they wrote the gcode for my Anycubic...

PR #5219 (2020-12-09): Add random function for gcode macros.
Adds a `random(min,max)` function for gcode macros. Fixes #4001
commit 003ccf38 (2020-12-09): Add a new macro function `random(min,max)` which will return a random number in the range specified by the parameters. The type of parameters defines the type of the output, if the inputs are doubles then result will be a double and if the inputs are int the output will be in. Uses C++11 random functionality, compiler compatibility on Linux should be checked. Fixes #4001.
closed 2020-12-09 by bubnikv