prusa3d/PrusaSlicer #4002: Double Retraction or Different Detration than Retraction

wagnbtd, 2020-04-05:
### Version
2.2.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
SeemeCNC_RostockMAX v3.2 - Bondtech Extruder - Duet2WIFI w/ Firmware v2.05.01

### Behavior
When printing, retractions seem to be different lengths than detractions. Seems to be at a  2 or 3 to 1 ratio, so that there is missing extrusion on the model. If the print section has several retractions back to back due to moves, the filament will retract to the point that nothing is extruding at all from the nozzle.

lukasmatena, 2020-04-06:
Could you please enclose a 3MF? There are several known issues with retraction.

wagnbtd, 2020-04-17:
Playing with settings more, I think this may be an extruder issue. For some reason, if I try to use more than 25mm/s detraction the motor doesn't want to drive the filament. I can retract at 60mm/s, but anything above 25mm/s detraction stalls the motor. I've tried multiple different vref settings.

rtyr, 2020-04-27:
> 
> 
> ... but anything above 25mm/s detraction stalls the motor. I've tried multiple different vref settings.

Closing. The issue is not related to PrusaSlicer.

closed 2020-04-27 by rtyr