prusa3d/PrusaSlicer #4003: The slicer does not seem to be able to parse some STL files correctly

artlav, 2020-04-06:
### Version
2.1.1

### Operating system type + version
Arch Linux, latest

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
The slicer does not seem to be able to parse some STL files correctly.
Attached is a simple model of a box with a hole through it, produced by CSG.

It is rendered perfectly fine by any other program i could try - Cura, Blender, OpenSCAD, KISSlicer, several online viewers.
In PrusaSlicer it produces a mess of polygons shown below.

[image]

What is expected:
[image]

### STL file
[hole.zip](https://github.com/prusa3d/PrusaSlicer/files/4435284/hole.zip)

artlav, 2020-04-06:
P.S. Also tried with PrusaSlicer-2.2.0+linux-x64-202003211856.AppImage, the behavior is identical.

foreachthing, 2020-04-06:
After running your STL through netfabb, it's all good!
[image]
(Highlighted was fixed with netfabb)
I'm on Win10 and with 2.2.0.

Blenders' 3D-Printing addon tells me that 44 edges are non-manifold and 3 faces intersecting.

artlav, 2020-04-06:
Interesting. So the model is bad somehow?

Why is it read properly by other slicers and software in that case?

madhephaestus, 2023-09-29:
This seems related to https://github.com/prusa3d/PrusaSlicer/issues/11199

closed 2026-09-29 by prusaslicer-bot[bot]