prusa3d/PrusaSlicer #4004: Mouse stuck in rotate mode (as if left button permanently pressed)

AtricoSoftware, 2020-04-06:
### Version
Version 2.2.0+linux-x64

### Operating system type + version
Ubuntu 18.04.4 (new install, no other software)

### 3D printer brand / version + firmware version (if known)
MK2, FW 13a, v3.2.3

### Behavior
* In "plater" tab, mouse movement rotates the bed as if left button were permanently depressed. Menu and tab buttons can be used but other buttons (move, rotate, etc) cannot be selected. Mouse wheel also works (zoom).  This was fixed temporarily by a reinstall but broke again after next reboot.  Mouse confirmed to be working correctly, all other apps/desktop features working normally.
* Reproduce: Open app, move mouse into main window
* Expected: Bed rotate only when mouse button pressed.
* Actual: Permanent rotate mode, app unusable.

_Feature request: No

#### Project File (.3MF) where problem occurs
No project file required to reproduce.  New project, add object, add stl all have no effect on problem.

AtricoSoftware, 2020-04-06:
UPDATE: Sorry, this looks like a ubuntu issue rather than slicer - Will continue to investigate

closed 2020-04-06 by AtricoSoftware

cowlicks, 2025-02-04:
This keeps happening to me, using gnome on arch linux