prusa3d/PrusaSlicer #4007: Filament Presets do not update afte running 'add filament' on drop down

touradnik9, 2020-04-06:
### Version
_Version of PrusaSlicer used goes here_
2.2.0+win64
[image]

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Still awaiting for the printer to arrive; it is an iMK3S MMU2S. It is a stock model
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_  On the plater page, when opening the filament dropdown, and selecting add filament, it will not refresh. I.e. even when you add a filament, the original dropdown of choices remains the same
* _Steps needed to reproduce the problem_
On the plater page, click one of the filament drop downs (I am using the MMU2 slicer version) and select a new filament. After you click finish, the page blinks slightly, like it is updating, but there is no change. 

On another note, when I open slicer this default object appears on the plater; not sure why, but it does not appear when you run generate gcode. See the photo below.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2020-04-06:
Which particular filament profile is not added? Please note, that only some of the filament profiles are compatible/visible with the MMU2 (multimaterial mode) printer profile. Try it with the "Original Prusa i3 MK3S MMU2S Single" printer profile instead.

touradnik9, 2020-04-06:
Hello,

I was trying to add a flex filament (flexfill98A), but it was not showing up; Why are only some filament profiles compatible / visible with the MMU2 multimaterial mode? What is the issue here?

Thank you for clarifying on the object that is the wipe tower.

Best,
Tim

rtyr, 2020-04-06:
That is intended. Flexfill 98A is not visible in multi-material mode. 
We are using completely separate filament profiles for MM printer profile. Filament temperature is set a bit lower for multi-material prints to improve reliability of the filament exchange sequence. There are also advanced parameters which are different for MMU1 and MMU2. We are publishing the profiles for filaments, which we tested in house with reliable results. Of course, you can experiment with your own filament profiles, ramming settings, etc.

rtyr, 2020-04-14:
Behavior explained. Closing.

closed 2020-04-14 by rtyr