prusa3d/PrusaSlicer #4008: Feature Request: Stacked copies

probonopd, 2020-04-06:
### Version

Any

### Operating system type + version

Any

### 3D printer brand / version + firmware version (if known)

Any

### Behavior

* As the user I want to print many copies of my object at once without having to remove prints from the build plate
* Use case example: Printing many COVID-19 face shields at once
* Example (credit: Devin Montes)

[image]

_Is this a new feature request?_ Yes

* Stack multiple copies of printable items over each other in z direction
* Each copy is exactly n layers (default = 2) apart from each other
* Check that the bottom surface of the object has the same shape as the top surface, in other words: ensure that the same n layer gap between the objects is there in each location
* And only add supports up to the bottom copy, do not add supports in between the copies

Description of the idea and explanation in this video by Devin Montes from MakeAnything: https://www.youtube.com/watch?v=MHKBvk8IAc0&t=715

bubnikv, 2020-10-15:
PrusaSlicer 2.3.0-alpha1 will allow printing with a gap. 
Supporting stacking is a very specific feature, because it depends on the quality of the input models. The only thing we can do in slicer is to adjust the Z gap, but then we will get a lot of bug reports if the stacking does not work due to the model issues.

probonopd, 2020-10-15:
> PrusaSlicer 2.3.0-alpha1 will allow printing with a gap.

What do you mean by this? (How to use it?)

bubnikv, 2020-10-16:
I mean that PrusaSlicer will no more stop printing a print with completely
empty layers, it will just emit a warning.

probonopd, 2020-10-16:
Thanks for the clarification. Since it appears that the space has to be fine-tuned depending on a large number of factors, I think this will have to do.

closed 2020-10-16 by probonopd