prusa3d/PrusaSlicer #4012: Copy and Paste stl from one instance of slicer to another

the-o-i-am, 2020-04-06:
### Version
2.2.0

### Operating system
Win 10

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3

### Behavior
Enable copy and paste of a model or model part between 2 PrusaSlicer instances so that model or model part can be copied from one instance and pasted into another
* _Expected Results_
Open 2 instances and load an stl into 1 instance
Click on Edit Copy
Goto other PrusaSlicer instance, click on edit and paste
* _Actual Results_
Copying from one instance does not register a paste in the other instance

_Is this a new feature request?_
New feature

NGatti1997, 2022-05-07:
Bump. Would be super helpful.

mepoland, 2022-12-12:
I could use this feature. I'd like all the modifiers and parts associated with an object in one instance, to be pasted over to the 2nd instance. This would save me the time of manually recreating those settings.

mtrakal, 2023-11-09:
This is must have, I open model from Printables.com open second model and want to print them together, but copy and paste is not working, instead it copy the same model from one instance of Prusa Slicer... 
So I must download STL model file, move them to slicer in which is already openned first model and than it's possible to print them together. Really slow down workflow, not comfortable at all.

mire1969, 2024-02-15:
Operating 3 Prusa Printer (SL1S, MK4 and XL) I would really appreciate this as a must have feature. Due to lack of this feature, currently I need to redo all print settings when switching printer by using only one instance, which is very inconvenient and not state of the art.

closed 2026-09-29 by prusaslicer-bot[bot]