prusa3d/PrusaSlicer #4014: GTK3 issues

Salamandar, 2020-04-07:
PrusaSlicer has different GUI issues, probably due to the (not stable) switch to gtk3.

One has already been reported : https://github.com/prusa3d/PrusaSlicer/issues/3571

Also, integer entries with "+/-" buttons are too small. A hard-coded width is probably set.
(In the screenshot, you can see part of the zone around the "+" button)
[image]

Lastly, the drop-down menus with icons seem to be too small.
[image]

AndKe, 2021-12-10:
what is the difference between GTK3 vs the "normal" .Appimage ?  - is the unmaled GTK2(very old)  - or GTK4 ?

bubnikv, 2021-12-10:
yes, the unnamed is GTK2.

AndKe, 2021-12-10:
@bubnikv Thank you, as a Ubuntu 21.10 user: which would you say I should use?

bubnikv, 2021-12-10:
Whichever suits you better.

AndKe, 2021-12-10:
I do not know what to look for to spot a difference... hence have no idea.
[image]

..thank you, have a great weekend. :)

Salamandar, 2021-12-12:
Use the GTK3 one :)

lukasmatena, 2023-07-10:
I believe this one was solved long ago. Closing.

closed 2023-07-10 by lukasmatena