prusa3d/PrusaSlicer #4016: Arrange puts objects too close for sequential printing

Patronics, 2020-04-07:
### Version

2.2.0+
### Operating system type + version
macOS Mojave, 10.14.6
### 3D printer brand / version + firmware version (if known)
Prusa i3 Mk3s MMU2, firmware version Not Applicable

### Behavior
* _Describe the problem_
When configuring the Slicer for sequential printing, (with the default radius and height of 45mm and 20mm respectively), the Arrange button does move the parts farther away from each other, but apparently not far enough, as I repeatedly receive the error message "Some objects are too close; your extruder will collide with them." until I manually move the parts a little farther away from each other.
* _Steps needed to reproduce the problem_

- Import a part into the slicer
- Check the "complete individual objects" checkbox in the Print Settings/Output options menu
- Return to the Platter view and press the Arrange button (or press Shift+A)
- Press "Slice"
- Get error message that some objects are too close.

* _Expected Results_
Arrange successfully spaces the objects out appropriately
* _Actual Results_
The error message described above until you manually move the parts farther away.
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]
_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[example_insufficent_spacing_error.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4446691/example_insufficent_spacing_error.3mf.zip)

braingame26, 2020-04-15:
I am also experiencing this issue.

bubnikv, 2020-11-21:
At least PrusaSlicer 2.3.0-alpha4 will allow configuration of the separation distance.

Mirarkitty, 2020-11-24:
I just got this issue in alpha4, but clicking again and it arranged it well for sequential printing. Very odd

tamasmeszaros, 2020-12-08:
This should now be fixed in 2.3.0 beta. Closing.

closed 2020-12-08 by tamasmeszaros