prusa3d/PrusaSlicer #4018: Allows Exporting of G-Code before slicing is complete

DRKMSTR, 2020-04-07:
### Version
2.2.0

### Operating system type + version
Windows 10 + 1909

### 3D printer brand / version + firmware version (if known)
Prusa Mini

### Behavior
During slicing, clicking on "Export G-Code" button can output a file name that doesn't contain the identifiers (Filament, Printer, Time)

bubnikv, 2020-04-08:
what do you mean by that?

foreachthing, 2020-04-08:
When clicking on "Slice now", the button changes to "Export GCODE". You can click it while it's still slicing. This is not really useful, then the user should not click it before it's done slicing (*cough*).

bubnikv, 2020-04-08:
> During slicing, clicking on "Export G-Code" button can output a file name that doesn't contain the identifiers (Filament, Printer, Time)

As far as I am known, the output file name contains placeholders for those variables, which are not known yet, such as the print time. These placeholders are filled in once the information is known. Therefore I believe everything works correctly.

bubnikv, 2020-11-24:
@foreachthing
> When clicking on "Slice now", the button changes to "Export GCODE". You can click it while it's still slicing. This is not really useful, then the user should not click it before it's done slicing (cough).

I suppose this is a matter of taste. If copying a large file on a removable media, you may start slicing, click G-code export to a slow SD card and go away from your computer to do your chores. I personally find it a useful functionality. In simple mode the application behaves differently.

closed 2026-09-29 by prusaslicer-bot[bot]