prusa3d/PrusaSlicer #4019: Feature Request - Dynamic Extrusion Width or something to reduce excessive travel moves

DRKMSTR, 2020-04-08:
### Version
2.2.0 

While printing some parts where the perimeters intersect (Face shields) I noticed that the stringing and blobbing was in areas I didn't expect. When I broke it down layer-by-layer, I noticed that a few areas had these intermittent lines where it would give the slicer the opportunity to leave the line and travel somewhere else. I thickened up the extrusion width for the upper layers using a modifier and removed the issue. 

Is there a way to incorporate a check (although it will likely take extra slicing time) to find these tiny extrusion lengths and see if a larger extrusion width can eliminate them?

### Operating system type + version
Windows 10 + 1909

### 3D printer brand / version + firmware version (if known)
Prusa Mk3s

### Behavior
This is a feature request. 

#### Pictures
This is with the standard "Extrusion Width"
[image]

This is the same layer with a larger "Extrusion Width"
[image]

Here are the .3mf files (I do apologize because the "un-modified" file is actually modified a bit already, just note that it's far far worse with the standard extrusion width)
[Example_ExtrusionWidth.zip](https://github.com/prusa3d/PrusaSlicer/files/4448002/Example_ExtrusionWidth.zip)

I hope this helps, it took me about 2 weeks to figure this out.

rtyr, 2020-04-08:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/2666, https://github.com/prusa3d/PrusaSlicer/issues/1806, https://github.com/prusa3d/PrusaSlicer/issues/1723, https://github.com/prusa3d/PrusaSlicer/issues/3822.

See this comment https://github.com/prusa3d/PrusaSlicer/issues/3822#issuecomment-598052997

dannydnl, 2020-04-08:
This needs to be fixed somehow.
Prusaslicer is doing way to many little extrusions. God forbid you try to print gears.
And yes, i tried a lot of combinations:
With and without Detect thin wall, different extrusion width and i ticked and unticked almost all ticks in the expert mode,  but nothing worked.
There is a serious need for, Minimum extrusion length and minimum extrusion width options in Prusaslicer
 I'm losing my mind. :-))
[image]
[image]
[image]
[image]

DRKMSTR, 2020-04-08:
@dannydnl 
You can try setting "Gap Fill Speed" to 0

According to this forum post: https://forum.prusaprinters.org/forum/prusaslicer/stop-gap-fill/

Hope that helps :)

yschroeder, 2020-05-03:
Perimeter configuration should work like this:

* Minimum number of perimeters
* Minimum shell thickness (in mm)
* Minimum allowed extrusion width (depends on you nozzle size)
* Maximum allowed extrusion width (also depends on nozzle size)

Then, PrusaSlicer should figure out on its own, how many perimeters of what width are needed to achive the best result which is avoiding gap filling.

I am also printing face shields and this is very annoying to get right.

DragonFinlay, 2022-12-08:
The issue is basically describing the definition of Arachne, yes. :)
I'll close it, thanks.

closed 2022-12-08 by DragonFinlay