prusa3d/PrusaSlicer #4020: moving parts after slicing

themay4547, 2020-04-08:
Version of PrusaSlicer used: 2.2.0+
Operating system type: mac os 10.13.6

I don't understand why but my parts move on right of bed after slicing. It's the same for all parts

[image]

[support_aimant_TouchMi.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4449218/support_aimant_TouchMi.3mf.zip)

Thanks for your help

rtyr, 2020-04-08:
It is caused by your start g-code.

It should be **G92 E0**.
[image]

Can we close this issue?

themay4547, 2020-04-08:
perfect, thank you beaucoup beaucoup ;-)

closed 2020-04-08 by rtyr