prusa3d/PrusaSlicer #4022: FEATURE REQUEST Infill and supports independent layer height

tixenre, 2020-04-08:
I'm doing some auto variable layer height prints. It is a pretty handle feature. But I think it would be a time-saver to set the infill and support layers height independently. 
Is it possible?? I know I can use combine infill every n layers, but the auto variable feature doesn't make it useful. 
I have seen this feature in simplify3d but not implemented with auto variable layer height

rtyr, 2021-02-12:
Supports are printed with variable layer height by default.

tixenre, 2021-02-12:
Yes! I know that supports are printed with variable layer height by default. 
What I asked for was to set the infill height independently from the height of the perimeter.

rtyr, 2021-02-12:
> 
> 
> What I asked for was to set the infill height independently from the height of the perimeter.

Which is what "combine infill every n" layers is for. This feature is used very rarely by the users as far as I know, so I don't really see a reason to extend it. It also introduces other related issues regarding cooling logic (min layer time), volumetric speed limits, etc.

Supports are always printed with independent layer height, regardless of layer height of the main object. Support layer height is synchronized with object layer height only when printing soluble supports (contact z 0)

tixenre, 2021-02-19:
Yes, I understand but in the case of using layer variable height would make a big difference in total print times.
For example:
> Print settings:
Layer height: 0.15mm
Combine infill every X layers: 2
If you use layer variable height in some parts and the print height ends up being 0.1mm, the infill will be printed at 0.2mm when you could, in theory, be printing at 0.3mm.
Same if you print with a bigger nozzle (which I do) and usually print with variable layers height between 0.1 and 0.4 mm. To have a constant infill height of 0.3 or 0.4 would mean to have an automatic "Combine infill every X layers".

Even better, you could, in theory, print perimeters at any variable height, but infill at 0.2, 0.25, 0.3, or whatever you decide. I know such of feature would cause problems with the infill anchor and perimeters in some shapes if the heights are not aligned. That's why I say in theory.
Even if what I propose is an advance/expert feature, "Combine infill every X layers" is also. 

I prepare .3mf  for living and I always, like many, try to find the best balance between better definitions and keeping times low as possible.  I think layer variable height and independent infill height would make a great combination of settings to achieve this.

Those are some thoughts, I'm aware of its physical limitations but I don't see why we couldn't have it in the slicer. Probably some developer will come up with some substantiation of why they don't or can't code it.  Meanwhile, I would cheer for this feature and if you find it useful, just don't use it.

closed 2026-09-29 by prusaslicer-bot[bot]