prusa3d/PrusaSlicer #4023: Add object serial number/revision when sliced

bcwadell, 2020-04-08:
### Version
2.2.0
2.2.0+win64

### Operating system type + version
Microsoft Windows 10 Home 10.0.18363 Build 18363

### 3D printer brand / version + firmware version (if known)
Prusa i3 mk3s fw 3.8.1-2869

### Behavior
The idea is to add a feature which embeds a text code into the print to assist with material, copy, and revision identification by end users.

When printing PPE and other iterative/repetitive prints it is helpful to be able to track copies and their settings.  For example, it is helpful to add a material code ("PLA", "PETG") to help those doing cleaning.  This can be added in the CAD but the file could be printed with the wrong material code still.  

When changing CAD files to iterate on design features it would help to be able to clearly track the sequence of versions.  While I can closely examine a design iteration and identify the specific changes it is difficult for an end user to describe to me which version they have.  

End users may wish to keep copies of the PPE object separated.  Embedding serial numbers would enable storage and reuse since users can then see if the PPE they are holding is the actual copy that belongs to them.  Embedding serial numbers also enables tracking throughout cleaning process.

This feature can also be seen to be useful in other scenarios where iterations and many copies are involved.  For example, when testing material properties of many copies.  For example, when testing different infill types or shells.  For example, when fit testing by printing multiple scaled versions.

desired behavior (for example):
* select stl name on right hand side
* right click, select Add Serial Number
* "Serial Number" appears in list of stls
* double click "Serial Number" in list to open a dialog
* dialog allows selection/entry of prefix, suffix, and alpha or numeric iterator format, material, scale
* for example the "Serial Number" object might generate the code "v2.2PETG" or "20200408_PLA" or "MGH 00001"
* click OK to approve choices
* when slicing, gcode is added to embed this Serial Number (SN) in the stl (and subsequently the gcode)
* SN can be selected and placed on print bed with usual move/rotate/scale tools including placement within an object
* slicing produces g-code with the serial number included
* the settings of the Serial Number object save as part of .3mf 
* subsequent slices bump the serial number per settings (bump meaning "MGH 00001" --> "MGH 00002"-->"MGH 00003" -->…)
* it's probably impossible, but ideally the gcode itself would bump the embedded code for each print copy

_Is this a new feature request?_
Yes

#### Project File (.3MF) where problem occurs
n/a

AStove, 2020-05-10:
Wouldn't that mean you'd have to reslice each time to bump the serial number? Semi-mass production just reuses the same gcode over and over again.

bcwadell, 2020-05-10:
Yes and no.  The feature request/idea has two parts to it.  The revision bump can be done at the time it is sliced and would be the same for all copies printed.  The serial number bump, as you point out, would have to happen after slicing.  
I can imagine how the second one (SN) might be done but I’m not an expert in limitations of g-code and whether it’s possible.  During slicing I would insert a g-code tag indicating the location, size, orientation of the SN label area.  Then in the g-code I would have a SN macro that tracks and bumps the SN which is triggered to insert the necessary g-code for the numbering at print time.  I think that may mean non-volatile storage access is needed.

SPNRmfg, 2022-04-25:
Could really use this feature, great request, well done.

westranm, 2022-11-17:
I print structural parts for my customers and would like traceability on the part itself back to the file I used to print it from Octroprint. I loosely do this by order number being placed into Octoprint file name, but it would be awesome if I could have a tool to place a serial number onto the print itself as requested by the closed feature requests (#6388, #6512, #6934) attached to this feature request (#4023). Using some kind of serial number tool, be able to place the serial # at minimum, on the bottom face of the object. Better, would be to allow an addition to any surface of the object, allowing option for either raised/depressed (emboss/deboss effects) text, depth of text, and dimension constraints.

dahappychappy, 2023-05-20:
Great idea, I am having troubles using the wifi upload. I change the model by a little in fusion. Reload the model from disk which has the same name. Then I slice it. The time it takes is exact same and it is telling me this is a duplicate filename. An incrementing versioning from prusa is needed.
Thank you

closed 2026-09-29 by prusaslicer-bot[bot]