prusa3d/PrusaSlicer #4027: Can't control the printing sequence of drawings in version PrusaSlicer 2.2 (was enable in v2.1)

Schrunch, 2020-04-09:
### Version
2.2.0+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Prusa MK2S, MK3, MK3S. The Printer and version doesn't matter

### Behavior
Actualy 2.2 decided the sequence so every gcode path is different for two drawings (or more) of the same object.
In version 2.1 the gcode was generated in the same order then it was displayed on the right list for each drawing for each slice height. I can't used the option of printing to complete individuals object because my drawings need to be one over the other.

Why did 2.2 is regressing to go back like version 2.0?

For the miniature parts I do and sell. I absolute need to have drawings printed in the same sequence I need.
For exemple, when superposing two drawing, if a line from the second drawing start at 90 degrees in the middle of another line on the first drawing, it will be ok but if a second line goes 90 degrees touching the end of the first one printed, it will be a disaster.

* _Steps needed to reproduce the problem_
  *Put two drawing and generate Gcode, change the order of the drawings and the path will change for each drawing.

* _Expected Results_

Same gcode path for each identical drawing and each object print in the order choose in the right list.
* _Actual Results_
No predictable path (travel)

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

It was an existing feature in version 2.1

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2020-04-09:
Starting with the PrusaSlicer 2.2.0-rc4 (release notes https://github.com/prusa3d/PrusaSlicer/releases), you can set sequential print order. Print order corresponds to the order in the object list, and the object list newly supports drag and drop to reorder objects. The sequential print order may also be indicated by labels in the 3D scene (visualisation of labels may be activated with the "E" key).

See this video for demonstration: https://youtu.be/wy3u4DNh_XQ?t=206 (3:22).

Does it help? Can we close the issue? If this doesn't solve your problem, please provide your 3MF project file.

[image]

rtyr, 2020-04-09:
Deleted my previous post. I misunderstood your description.

bubnikv, 2020-04-09:
Please provide drawings. It is not clear what you want our slicer to do.

Schrunch, 2020-04-09:
Those gcodes where produce from the same 3mf project.
 
The code from V2.2_2Frame_0.15mm_PLA_MK3.gcode  prints in this order: Lines.stl,  Frame.stl, Lines-copie-copie.stl,  Frame-copie-copie.stl, , Lines -copie.stl,  Frame -copie.stl
The code from V2.1_Frame_0.15mm_PLA_MK3.gcode  prints in the same order has shown in the list at the right: Frame.stl, Lines.stl, , Frame -copie.stl,  Lines -copie.stl,  Frame-copie-copie.stl,  Lines-copie-copie.stl
 
Those exemple are not critical for printing with good result but for the more complex miniature things I do using a 0.15mm nozzle it is a must being able to control orders and layers(it why I can’t use the complete individual objects option).
 
Sorry, I was unable to drop files into my request.
 
Thank you,

bubnikv, 2020-11-21:
Sorry no drawing provided. Anyways, objects are printed in the order they are listed in the object list. When pressing the "e" hotkey, a label with the order indicator will be drawn in the scene.

Schrunch, 2020-11-21:
Thank you, it works find.

bubnikv, 2020-11-21:
> Thank you, it works find.

Would you please close your issue then?

closed 2020-11-21 by Schrunch