prusa3d/PrusaSlicer #4029: Slicing error with

jodysankey, 2020-04-09:
### Version
Version 2.2.20

### Operating system type + version
Ubuntu 18.04

### 3D printer brand / version + firmware version (if known)
Prusa Mk3S

### Behavior
Fairly simple STL from openSCAD (4648 facets, 1 shell, is manifold) fails to slice in some orientations with an error "empty layers detected". In other orientations the slice succeeds with errors (parts of the print appear "inside out"). The part looks fine in the 3D editor view.

Importing the STL file into blender and re-exporting leads to a smaller STL file but produces the same result

#### Project File (.3MF) where problem occurs

[bug_demo.3mf.tar.gz](https://github.com/prusa3d/PrusaSlicer/files/4454061/bug_demo.3mf.tar.gz)
[image]

rtyr, 2020-04-09:
I am unable to reproduce it on 2.2.0-win64.

[image]

foreachthing, 2020-04-09:
Slicing your 3mf with the latest build (https://github.com/prusa3d/PrusaSlicer/commit/3e15d22f35ac94b4859be5fe4535122a162f6695) looks all good:
[image]

Some errors detected: 8 degenerated facets; 8 faces removed.
If you already use Blender, try the 3D-Printing addon and fix your mesh with it. It might help.

FidelCapo, 2021-02-01:
I can't test the fix for this model because there is no stl file available, only a 3mf file. However, according to tests of other models, this issue should be corrected in next release.
Closing.

foreachthing, 2021-02-01:
@FidelCapo, FYI: if you unzip the 3mf file, you can reach for the stl ;-)

FidelCapo, 2021-02-01:
@foreachthing There is only .model file when 3mf is unzipped..
@jodysankey  Can you please attach STL file? Thank you.

foreachthing, 2021-02-01:
Here's the STL from the 3mf file. Still has some errors in it...
[cable_hook_v1.zip](https://github.com/prusa3d/PrusaSlicer/files/5904219/cable_hook_v1.zip)

jodysankey, 2021-02-03:
I moved on and recreated the model using different geometry, but I think the attached is the STL I was using at the time - probably the same as @foreachthing extracted.

Trying again in 2.3.0 shows some interesting errors such as "mm3_per_mm() produced negative flow."

[cable_hook_v1.tar.gz](https://github.com/prusa3d/PrusaSlicer/files/5917529/cable_hook_v1.tar.gz)

rtyr, 2021-06-30:
Should be fixed with https://github.com/prusa3d/PrusaSlicer/commit/9c5dd358ab23e9351dce0b9c71eaa5ddf92d6f17. Closing.

closed 2021-06-30 by rtyr