prusa3d/PrusaSlicer #4033: Sequential Printing does not work on objects split into parts

tillin9, 2020-04-09:
PrusaSliver: 2.2.0+linux-x64 (Seems to happen on all versions I had handy 2.0.0, 2.1.0, 2.2.0+beta) and is easily reproducible. 

Import an .stl with multiple component objects. HeartGears5-6.stl from https://www.thingiverse.com/thing:243278 was what I was using to repro.

Select Split into Parts to turn the single object into component objects.

Changing Print Settings > Output options > Complete individual objects will not give the desired goal of printing the component objects one by one. 

If a 2nd copy of the .stl is imported, split into parts, and duplicate parts are deleted, this will work as intended. 

The lack of the preview going object by object (even when the resulting gcode is good) makes it harder to see what is going on. 

While not a critical issue (it is possible to work around), this was a tad annoying, so hoping the fix gets into a future release.

rtyr, 2020-04-09:
Maybe I am missing something, but I don't see an issue here. You should use "Split to objects" option for this purpose.
Split to parts = 1 object with 17 parts
Split to objects = 17 objects which can be printed sequentialy using "Complete individual objects".

Individual objects are printed in the order according to object list on the right side and you can even visualize that order on the plater by using "E" key. This is implemented since PS 2.2.0-rc4.

tillin9, 2020-04-10:
You are correct, sorry about that.

closed 2020-04-10 by tillin9