prusa3d/PrusaSlicer #4034: Blue Screen of Death - Random - Kernel Error

ec-tune, 2020-04-10:
### Version
_Version of PrusaSlicer used goes here_
PrusaSlicer-2.2.0+win64-202003211145

_Use `About->About PrusaSlicer` for release versions_
Version 2.2.0+win64
_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 Home 64 bit 
Version 1909 (OS Build 18363.720)

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
In attached zip file

### 3D printer brand / version + firmware version (if known)
Ender 3 with Melzi 1.1.4 - stock firmware

_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Ender 3, factory stock firmware for Melzi 1.1.4 board
### Behavior
* _Describe the problem_
Random blue screen of death. Completely re-installed windows on laptop prior to blue screen of death. Basis programs reloaded to 3D print. 

* _Steps needed to reproduce the problem_
Unknown - random
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_
No
#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
Windows minidump, Windbg output for minidump, and Prusa Slicer System Information contained in
[4_9_2020_Windbg_MiniDump_BSOD.zip](https://github.com/prusa3d/PrusaSlicer/files/4459645/4_9_2020_Windbg_MiniDump_BSOD.zip)
 Zip file.

bubnikv, 2020-04-10:
> Random blue screen of death. 

Is your system crashing? If so, then it is not PrusaSlicer's fault. You may have windows installation issue, driver issue (likely a graphics card issue) or hardware issue (possibly a memory module failure).

ec-tune, 2020-04-10:
Yes, BSOD crash, but only crashes when using Prusa Slicer. Gave a kernel security error before rebooting.
Brand new windows installation, Intel 8th gen Core I3 with Intel on-board graphics, Intel 620 series graphics.  All drivers up to date. Going to see if it repeats.

ec-tune, 2020-04-14:
Wiped drive and reinstalled Windows 10 once again. Was getting BSOD on Windows installation with Lenovo USB installation media. 
Used Microsoft USB installation media creator and install went OK. Only one BSOD before a driver updated. Just reinstalled Prusa Slicer software. Will update this report if necessary.

rtyr, 2020-04-15:
I believe we can close this. This is clearly some issue on your system.

closed 2020-04-15 by rtyr