prusa3d/PrusaSlicer #4035: UI Issue in the French version

olikaf, 2020-04-10:
### Version
2.2.0 win 64

### Operating system type + version
Windows 10.0.18362 Numéro 18362

### 3D printer brand / version + firmware version (if known)
Not a printer issue

### Behavior
There is an issue in the French version of PrusaSlicer
The text "Facteur de redimensionnement" is too long, and the icons ( reinit rotation,.. )  on the right are almost unreachable

[image]

JohnnyDeer, 2020-04-14:
Hi, issue is reproduced and confirmed, thank you for reporting.

YuSanka, 2020-04-16:
We discussed this situation with our French translators and they changed the phrase to shorter one.
[image]

This fix will be a part of the next release. Close.

closed 2020-04-16 by YuSanka