prusa3d/PrusaSlicer #4036: Feature requests: SL1 supports

nigeljohnson73, 2020-04-10:
I'm new to the SL1 and love it so far, but the model prep is a mission, I spend literally hours tweaking the supports for fine wire-based models I have... So a few tweaks to the software would be really nice. Should I break these out into individual requests?

1) Auto generate island only. This would save hours of prep. Only doing the supports that are 'required' then let me tweak those, then adding other supports manually. I think I found a work around, to auto-generate, lock under islands that select all and delete... seems like an ok workaround.

2) Auto generate ignoring hollow. Oh my, you do like a lot of supports in there don't you :D

3) Multiple select by dragging the mouse with the button clicked... Again, loads of time saving being able to delete a large number of support points.

4) Re-routing support poles - if I have a fine voronoi model with wires that are 0.4mm in diameter, it seems Prusa Slicer sometimes has trouble seeing them and can put a pole right through a wire. The ability to select a pole and add turns would allow me to route the support through the wires.

AndrewFarley, 2020-05-01:
OMG #2 Please... when hollowing there's an ungodly amount of supports placed inside the model.  Holy hell, I find myself going in and manually removing most of the internal points.

And #3 would definitely help a lot since it is definitely a mission when wanting to disable a bunch of them in an area.

lukasmatena, 2020-05-01:
@AndrewFarley 
Generation of supports in cavities will hopefully be improved in the next release.

> Multiple select by dragging the mouse with the button clicked... 

That is indeed helpful. And for that reason, it has been implemented in the very first version of the support points editor. Just Shift + drag. Click the "?" icon in the dialog to see complete list of shortcuts.

AndrewFarley, 2020-05-07:
> That is indeed helpful. And for that reason, it has been implemented in the very first version of the support points editor. Just Shift + drag. Click the "?" icon in the dialog to see complete list of shortcuts.

@lukasmatena This functionality does not work at all, or very poorly.  I just tried on three different models, the shift drag I see the selector come out, and then if I'm lucky it grabs one or two points, amongst the 100 or so that I selected over.  I can send a video if it helps

Area5142, 2020-06-21:
In support placement tool dialog I would like a feature to hightlight all islands to make it easier to manually place support the right places. It could be combined with a threshold angle to highlight overhangs that needs support too.

Complex detailed organic and voronoi models is difficult to give the right support without this type of help.

lukasmatena, 2020-06-23:
@AndrewFarley Sorry, I missed your reply. I have seen a model with similar problem, the issue was that it had uneven surface with lots of bumps on it and the support points were often hidden behind them (the rectangle uses center of the ball for reference, so if the center is obscured, the point is missed). Is your case similar?

@Area5142 There is a workaround - generate automatic supports, switch to manual editing mode and check 'Lock supports under new islands'. Then edit the supports as needed. As for the slope visualization, try enabling View->Options->Show slope if that's helpful for you.

AndrewFarley, 2020-06-23:
@lukasmatena I mean... I can't tell you exactly, but typically I'm trying to remove the insane number of supports that it tries to place inside the hollowing.  So, hollow and generate supports for a model, and then I try to remove the points of the model while I'm using the Clipping feature to hide half the model so I can try to drag select a bunch of points inside.  I wonder/suspect that the clipping feature and the drag-select do not mix well.  Because I'm 100% always using clipping when I'm trying to slice.

It's so frustrating the insane amount of unremovable supports that are placed inside the hollowing.  I haven't looked at the latest code, but I'm hoping that they have some feature/checkbox to disable generating supports inside the hollowed area.  In the case of certain complex models that need it, I'd rather add them internally myself.

lukasmatena, 2020-06-23:
>  I wonder/suspect that the clipping feature and the drag-select do not mix well.

Thanks, I'll focus on that when I try to provoke the problem. It is possible that there is/was some bug related to that.

> t's so frustrating the insane amount of unremovable supports that are placed inside the hollowing.

I share your frustration. This is a separate problem and I hope we'll fix it before 2.3.0 release is out.

Area5142, 2020-06-23:
@lukasmatena I've tried both tools and they are of little help when making supports for a voronoi or sililar model with lots of tiny details. A more traditional/technical model with large surfaces is easier to support.
_Lock supports under new islands_: The name is confusing as it locks already placed support under island, not new island (Am I wrong?). It will not tell me, where the islands or low angle areas is or places that still needs support. 
_View -> Options -> Show slope -D_: As I understand this tool, it will show areas from 0 to red set point in red and areas to the next set point in another colour. This is great for a model with large areas where the colouring is easy to spot. If the model has fine and uneven surface it is hard/impossible to spot where the areas is. If support is already generated it will block the view and I have to go to _Manual support_ to turn off the showing of support (don't know other way).
The _Show slope_ function would be great if the colours where more visible in a contrast colour or preferably the rest of the model could be turned off and only the parts of the model with selected angles where displayed.

lukasmatena, 2020-06-23:
@Area5142 
> it locks already placed support under island, not new island (Am I wrong?)

You are not wrong. It only prevents supports under new islands to be moved or deleted. But autogeneration should never miss a new island, so all new islands should be supported by the "blue" points and none should remain so in a way, it shows you where they are. I agree that it is not optimal from UI point of view, but realtime support calculator is significantly more difficult to implement with uncertain results.

>  It will not tell me, where the [...] low angle areas is or places that still needs support.

No, it will not. The software does not even know where to put supports. It can only do a guess which is what autogeneration is for. If you do manual editing, you have already expressed that the guess from the software is not acceptable to you, so you should not trust hints about what "needs support" (if there were some). The only hint that is objective is the slope of individual facets and the information about new islands. Those is covered by "Show slope" and the "Lock support points..." functions, respectively.

> If support is already generated it will block the view

The problem of supports getting in the view can partially be solved by the clipping plane (bottom slider in the dialog, shortcuts R and Ctrl+Mousewheel).

Don't get me wrong, I agree that this could be improved, if/when there is time to do that. I'm just describing current state.

closed 2026-09-29 by prusaslicer-bot[bot]