prusa3d/PrusaSlicer #4037: Feature request: 3mf profiles cleanup

nigeljohnson73, 2020-04-10:
Now that there is a lot more work involved in SL1 printing, I'm saving more projects to revisit after printing. I'm finding that if I change materials between loading the project, a new set of material and print settings are created for the project... and then they don't go away unless I go into each tab, and delete them myself.

What would be really nice is if these settings were loaded into a temporary list, then, when you close the project, open a new one... load Prusa Slicer after a crash... whatever, that the temporary area is wiped so you don't end up cluttering the profile lists

rtyr, 2020-04-10:
These "imported" profiles are only temporary, so you don't have to delete anything, they will be gone after PS restart. It happens only if settings included in the 3MF project file doesn't match to system profiles.
You can also open 3MF file without settings using "Add.." button, File-Import-Import 3MF or CTRL+I.

Can we close this issue?

nigeljohnson73, 2020-04-10:
Ok, mine hung around through a couple of restarts. but I'll give it a try when I change material again. Happy to close and I'll shout if it happens again.

closed 2020-04-10 by nigeljohnson73