prusa3d/PrusaSlicer #4038: Thin wall slicing bug?

thierryzoller, 2020-04-10:
### Version
Latest -2.2

### Operating system type + version
Sidewinder X1

### Behavior
Slicer slices certain layers in a weird way (see screenshots)
The line is actually straight all the way up, not sure why it is slice artifacts.

[image]
[image]

[Prusa_Bug_Report1.zip](https://github.com/prusa3d/PrusaSlicer/files/4461292/Prusa_Bug_Report1.zip)

thierryzoller, 2020-04-10:
Layers 151 and 152 present the artifacts

rtyr, 2020-04-10:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/149, https://github.com/prusa3d/PrusaSlicer/issues/227, etc.

It is related to the "detect thin walls" parameter. Disable it and set extrusion width of external perimeters to 0.4mm. It should fix the issue with this particular object.

thierryzoller, 2020-04-10:
Now I have this, also see #4040 

[image]

rtyr, 2020-04-10:
This is actually wiping movement ("wipe while retracting" parameter). For some reason, you are using completely different printer profile now (Prusa_Bug_Report1.zip vs Prusa_Bug_Report2.zip).

rtyr, 2020-04-17:
Closing. Duplicate of #149, #227, etc.

closed 2020-04-17 by rtyr