prusa3d/PrusaSlicer #4040: Overhand Perimeter detection Error ?

thierryzoller, 2020-04-10:
### Version
2.2 Latest

Detects Overhangs where there are none. Result is a print with the left side of this perimeter only half printed.

[image]
[Prusa_Bug_Report2.zip](https://github.com/prusa3d/PrusaSlicer/files/4462373/Prusa_Bug_Report2.zip)

rtyr, 2020-04-10:
I am unable to reproduce it with your 3MF.

[image]

thierryzoller, 2020-04-10:
Looks like you havent enabled Travel

rtyr, 2020-04-10:
Oh, now I understand what did you meant. Those blue lines are travel moves, not overhangs. So I guess we can close this issue.

closed 2020-04-10 by rtyr

thierryzoller, 2020-04-10:
It doesnt travel the full line - like I said I get a half triangle shape upper piece.

rtyr, 2020-04-10:
Let's leave it in 4038. This is not related to overhangs.