prusa3d/PrusaSlicer #4041: Feature Request: Generate Delamination Layer for Separating Stacked Parts

jason0l, 2020-04-10:
It would be great if PrusaSlicer can generate a specified layer where the hotend can do a controlled drop in temperature to a temperature at which the layers delaminate. Controlled delamination layer would really help with separating parts that are printed stacked
[image]

rtyr, 2020-04-10:
I believe this is already implemented, because you can easily add custom g-code to the specified layer. For example M104/M109 for temperature, M106 for fan speed, M221 for flow, etc..

[image]

jason0l, 2020-04-11:
Thank you. Didn't know I can enter g-code like that. 
It would be preferable though to have something a bit more automated that can insert temperature changes early on so that once it starts the delamination layer, the hotend is already on its way reaching the lower target temperature. Same case for when getting back up to temperature. Gradual adjustments to the speed can also compensate for changing temperature in normal printing layers

jason0l, 2020-04-11:
Although, an issue with PrusaSlicer is that it doesn't like gaps in an entire layer, and it always sees two separate models stacked on top of each other with no separation as one model

bubnikv, 2020-10-15:
The gaps will be allowed in PrusaSlicer 2.3.0-alpha1, the error will turn to notification.

closed 2026-09-29 by prusaslicer-bot[bot]