prusa3d/PrusaSlicer #4044: First layer settings in PrusaSlicer are ignored

kashalkar, 2020-04-11:
### Version
2.2.0

### Operating system type + version
OSX High Sierra

### 3D printer brand / version + firmware version (if known)
MK3s with latest firmware

### Behavior
I find that printing first layer with Speed=50% gives me a perfect first layer.
I can change speed down to 50% using the orange dial on the mk3s as soon as the print starts. 

However, when I set first layer speed to 50% in prusaslicer. It is ignored by the MK3s. The printer runs at 100%.

Expected behaviour is that printer should print first layer at 50% and then go upto 100%.

rtyr, 2020-04-11:
Percentage used in the slicer is related to default speeds set in the print profile. Feedrate percentage on the printer is completely different thing managed by printer firmware. Default speed for first layer is currently set to 20mm/s in our print profiles, so if you want 50% speed of that, set it to 10mm/s. 

Closing.

[image]

closed 2020-04-11 by rtyr

readysetawesome, 2020-11-25:
I'm having the same issue and I don't understand the response above. I can set 20mm/sec exactly for first layer speed and it is ignored every time I slice. It continues to use 40mm/sec regardless of what I set.

rtyr, 2020-11-25:
I can look at your settings if you provide your 3mf project file.

readysetawesome, 2020-11-26:
Thanks, I think I misunderstood what was happening because of how the printer displays print speed.

dumpguy, 2020-11-29:
Hello to all,
is it possible that the first layer speed setting is not applied to brim and supports?
I had permanent problems with the adhesion of the first layer of these structures, while the first layer of perimeter works flawlessly.
Since reducing the general speed setting of brim and supports it´s fine. But if the first_layer_speed (I use the standard 30 mm/sec) would have been applied as it is for the perimeter, this step wouldn´t be neccessary, right?

Regards,
_Stefan_