prusa3d/PrusaSlicer #4047: Print in place parts

bman2000, 2020-04-11:
2.2 final release

When I import two parts that are to be printed together such that one part pivots within the other with a .8mm clearance, they print as one.

I have changed the clearance from 0.2mm to 0.8mm without an improvement.

I am using 0.1mm layer height and a 0.3mm nozzle with low.

The preview shows the correct slicing but it does not print that way.

rtyr, 2020-04-11:
Please provide 3MF.

bman2000, 2020-04-12:
Here is the file. This part prints ok using cura 4.5. I am not sure the
settings are the same though.

BR

Bernie

rtyr, 2020-04-12:
Where? I don't see any file :-)

bman2000, 2020-04-12:
Here it is again

rtyr, 2020-04-12:
Still nothing. Use zip archive.

bman2000, 2020-04-12:
ok lets try this.

rtyr, 2020-04-12:
Nope. Just upload it directly using the web browser and check the result...

bman2000, 2020-04-12:
Just added it to my original post

rtyr, 2020-04-12:
I checked your 3MF and generated gcode. I can see two potential issues.

The issue you are experiencing might be caused by "Detect bridging perimeters" parameter, which you have enabled in your settings. With this setting, some of the perimeters in the "pivot area" are printed with the bridge flow, which means it will be printed with significantly thicker layer (based on the nozzle diameter, 0.3mm in this case). Try to disable this parameter.

Second issue I noticed is with the model itself. It should be fixed before printing to avoid gaps (which are present in the Cura too). Right click on the model and select "Fix through the Nefabb" option.  

[image]

bman2000, 2020-04-13:
Tried your suggestions but saw no improvement. The parts are still welded.

Betnie

rtyr, 2020-04-13:
I checked generated gcode and it is printed separately as expected and as previewed by PrusaSlicer.

I tried to print it on the MK3S. No problem. Only thing preventing free movement was brim.

I'm sorry, but you will probably have to tune up your profiles or your printer. I don't see an issue on the PrusaSlicer side here. If your printer produces better result with the Cura, you have probably better tuned profiles there. 

See attached photos.

Your gcode previewed by Simplify3D/CraftWare:
[image]

Printed part from Prusa i3 MK3S with similar settings as possible. (0.4mm nozzle).
[image]

closed 2020-06-14 by rtyr