prusa3d/PrusaSlicer #4051: [Feature Request] Add Optimise Orientation to command line

Maximo1491, 2020-04-12:
### Version
PrusaSlicer 2.2.0

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Resin Printers

### Behavior
Been looking to write a script for Resin models that can automatically import the model, auto-orientate, then auto-support, then export the stl inc supports so I can set a batch of files going overnight. I believe I can do the auto support and auto-export already but can't see anything to do the auto-orientate. Would really help improve productivity!

rgauny, 2021-05-23:
@Maximo1491 can you show an example of the steps you use to do the repair, auto-support and auto-export?

Maximo1491, 2021-05-27:
I haven't implemented it yet as the auto-orientate was key to what I wanted to achieve, but looking at the examples on the Slic3r page and then modifying it with the prusa options, I assume something like ".\prusa-slicer-console.exe --export-stl --repair --scale 100% --printer-technology SLA --supports-enable" would work?

closed 2026-09-29 by prusaslicer-bot[bot]