prusa3d/PrusaSlicer #4053: Stange behavior of the Pattern Spacing

SteWhite, 2020-04-12:
### Version
2.2.0
### Operating system type + version
Windows 10 pro 64bit
### 3D printer brand / version + firmware version (if known)
Anycubic i3 mega-s
### Behavior
If I change the pattern spacing, a part of the model is no supported.

_Is this a new feature request?_
no
[image]
[image]

#### Project File (.3MF) where problem occurs
[supporti errati.zip](https://github.com/prusa3d/PrusaSlicer/files/4466786/supporti.errati.zip)

neophyl, 2020-04-12:
The support is supporting the overhang perimeters.  Once the perimeters are laid down the other interior area is printed using bridging so doesn't need to be supported.  When the spacing is smaller it has more 'definition' so it can follow the unsupported perimeter, when larger it treats the whole area as one section.   
Seems to be doing exactly what I'd expect.

SteWhite, 2020-04-12:
Thanks for the reply.
(the orientation is for the example)
With 2.5 I don't have that any hole/bridge.
With 2 i have 8mm hole/bridge
It's inconceivable. I have never seen Simplify3D think in this way.
I print often with nozzle 0.25 and layer 0.05 and I cannot cover up a 8mm bridge.
Mayby the MK3s can (in few weeks I will find out) but with anycubic it's impossible.
[image]

rtyr, 2020-04-12:
> 
> 
>  and I cannot cover up a 8mm bridge.

Disable "Don't support bridges" setting. It should solve your issue.

[image]

SteWhite, 2020-04-12:
ok perfect!! Sorry, I'm blind.
Do you know how can I reduce the extra inflation distance? With "support enforcer" it's the same.
Thanks a lot!
[image]

rtyr, 2020-04-12:
There is no special setting for extra inflation distance. I guess you can avoid this by different pattern spacing or by increasing XY separation.

SteWhite, 2020-04-12:
yes, I have tried that features, It partially solves the problem. Thanks a lot.

closed 2020-04-12 by SteWhite