prusa3d/PrusaSlicer #4058: Paneli with layer slider is too thin

ochm, 2020-04-13:
### Version
2.2.0+Win64

### Operating system type + version
Win10

### Behavior
Please, can you add add two or more pixels to layer slider section for better readibility of numbers?

[image]
[image]

foreachthing, 2020-04-13:
Or adaptive width or rotate label 90deg. ;-)

JohnnyDeer, 2020-04-14:
Hi, i can see this issue when slicing in spiral vase mode. Do you have any more ways to reproduce it?

ochm, 2020-04-24:
Yes, the issue is visible when number of layers is very high, like in vase mode.

rtyr, 2021-02-12:
Should be fixed in 2.3.0. Closing.

closed 2021-02-12 by rtyr