prusa3d/PrusaSlicer #4060: [bug] reproducible hard-crash | when using layer height modifier

BeldrothTheGold, 2020-04-13:
### Version
latest release (2.2.0) and latest commit ebcc8ef7c21eb04fbb9d935969485de58d9a11b8 

### Operating system type + version
Microsoft Windows [Version 10.0.18363.720]

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Reproducable hard crash when using attempting to slice the attached 3MF. 
Issue appears to be caused by layer height modifier.

Looks to be caused by an out-of-bounds vector access here:

```	// slicing in parallel
	std::vector<ExPolygons> this_slices = this->slice_volume(slice_zs, ranges, SlicingMode::Regular, *model_volume);
	if (out.empty()) {
		out = std::move(this_slices);
		merge.assign(out.size(), false);
	} else {
		for (size_t i = 0; i < out.size(); ++ i)
			if (! this_slices[i].empty()) {     <----- crash
				if (! out[i].empty()) { 
					append(out[i], this_slices[i]);
					merge[i] = true;
				} else
					out[i] = std::move(this_slices[i]);
			}
	}
```

Exception thrown at 0x00007FFD4BE095F8 (PrusaSlicer.dll) in prusa-slicer-console.exe: 0xC0000005: Access violation reading location 0x0000000000000000. occurred

>	PrusaSlicer.dll!Slic3r::PrintObject::slice_modifiers(unsigned __int64 region_id, const std::vector<float,std::allocator<float>> & slice_zs) Line 1998	C++
 	PrusaSlicer.dll!Slic3r::PrintObject::_slice(const std::vector<double,std::allocator<double>> & layer_height_profile) Line 1765	C++
 	PrusaSlicer.dll!Slic3r::PrintObject::slice() Line 113	C++
 	PrusaSlicer.dll!Slic3r::PrintObject::make_perimeters() Line 149	C++
 	PrusaSlicer.dll!Slic3r::Print::process() Line 1581	C++
 	PrusaSlicer.dll!Slic3r::BackgroundSlicingProcess::process_fff() Line 89	C++
 	PrusaSlicer.dll!Slic3r::BackgroundSlicingProcess::thread_proc() Line 202	C++
 	PrusaSlicer.dll!Slic3r::BackgroundSlicingProcess::thread_proc_safe() Line 244	C++
 	[External Code]	

#### Project File (.3MF) where problem occurs
[headphone_holder_more_printable_ASA.zip](https://github.com/prusa3d/PrusaSlicer/files/4470161/headphone_holder_more_printable_ASA.zip)

ochm, 2020-04-13:
Yes, crashes reliably

hejllukas, 2020-12-09:
It should be fixed with https://github.com/prusa3d/PrusaSlicer/commit/bb2eecb42c4cdae4e72d6dd1c6a7ee5546f648bf.

closed 2020-12-09 by hejllukas