prusa3d/PrusaSlicer #4061: Hard crash when launching. PS 2.2.0 and 2.2.0 RC

makro50, 2020-04-13:
### Version
2.2.0 2.2.0RC

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 64B

### 3D printer brand / version + firmware version (if known)
i3 Mk3s adn Ender 3 Pro
### Behavior
Crash on start, before it would crash on slice, I was able to get around this by importing a basic model, slicing then deleting and importing the model I wanted to slice.

_Is this a new feature request?_
No

makro50, 2020-04-13:
Both printers are running on octopi. Seems that when Ender printer is printing, PS cannot open.

keawepoo, 2020-04-14:
Im having the same issue, I know that if you go into your AppData>Roaming> and remove PrusaSlicer itll start the program after a few times at least.

rtyr, 2021-02-12:
Is this issue still valid? Does it happen with 2.3.0?

keawepoo, 2021-02-12:
Unknown for me... I haven't had a chance to try it out yet. @makro50  what about you?

rtyr, 2021-07-23:
No response. Closing.

closed 2021-07-23 by rtyr