prusa3d/PrusaSlicer #4062: Add possibility to ignore "Empty layer" error

espr14, 2020-04-14:
v2.2.0 win64

### Motivation
I want to print objects one on other and, after print, I want to break them apart easily. That means leaving like 0.4 mm gap (in z axis) between them.

### Issue
Empty layer brings error message and the slicing is stopped.

### Possible solution
Add "Ignore" button (e.g. only in expert mode) to continue with slicing.

iamnotahippy, 2020-04-22:
I'd like to add some urgency to this feature request, so objects such as these stacked face shield covers can be sliced. https://ucsf.app.box.com/s/u1t7dlj20tjk6k7borpae9hor24otx3j

cfelicio, 2020-05-21:
Ok I'm also experiencing this issue and since this has no attention from the devs (I'm also trying to print face masks), here is where the problem is:

Line 639 -> PrusaSlicer-version_2.2.0\src\libslic3r\GCode.cpp

 if (has_extrusions && layer_to_print.print_z() > maximal_print_z + 2. * EPSILON)
                throw std::runtime_error(_(L("Empty layers detected, the output would not be printable.")) + "\n\n" +
                    _(L("Object name")) + ": " + object.model_object()->name + "\n" + _(L("Print z")) + ": " +
                    std::to_string(layers_to_print.back().print_z()) + "\n\n" + _(L("This is "
                    "usually caused by negligibly small extrusions or by a faulty model. Try to repair "
                    "the model or change its orientation on the bed.")));

I will try to re-compile with this commented out. If anyone wants I can post a compiled version as well..

bubnikv, 2020-10-15:
Implemented as a notification in PrusaSlicer 2.3.0-alpha1.

closed 2020-10-15 by bubnikv

4wrxb, 2021-04-26:
@bubnikv is this in v2.3.1? I am getting an toast that says "error" and my exported GCODE is the slice that was created before I added an un-printed object to block brims.
[image]

4wrxb, 2021-04-26:
FYI, as a work-around merge your zero-printed objects with a hack-printed object (set that object to 0 as well, except for 1 base layer). If the hack object is (partially) outside of the zero-printed object it will be treated as a first layer for the whole object. You want this away from your part so that the hack object doesn't intersect.

Keep-out plus hack object (hack object selected)
[image]

Sliced brim around keep-out plus some base-layer of the hack object.
[image]