prusa3d/PrusaSlicer #4063: Orientation affects. hole drilling

nigeljohnson73, 2020-04-14:
### Version
2.2.0+

### Operating system type + version
MacOS 10.14.6 Mojave

### 3D printer brand / version + firmware version (if known)
SL1 running 1.3.3

### Behavior
Hole drilling is affected by model orientation. A model that drills and hollows correctly when imported (flat on the bed) can fail later.

[image]

[image]

* Import the attached STL [_for Prusa.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/4474527/_for.Prusa.stl.zip)
* Note: no geometry errors are reported.
* Hollow to the parameters above and place a hole roughly central in the base
* Auto generate supports - all good
* Rotate model 19 degrees in Y
* Auto generate supports - all good
* Rotate model 1 degree in Y (Total now 20)
* Auto generate supports - failure

Also tried different-sized holes and different hole locations on the base. Still failure. Run the model through netfabb online in case a geometry problem Prusa Slicer doesn't detect and still failure.

Expected results would be that the hole and hollowing occur when the model is flat to get a new geometry and then the translation would occur and finally support placement.

#### Project File (.3MF) where problem occurs
[_for Prusa.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4474526/_for.Prusa.3mf.zip)

taarraas, 2020-05-10:
I have the same problem. Have you found workaround?

bubnikv, 2021-09-22:
It may happen that a nearly degenerate mesh degenerated during the rotation or scaling operation: Two coordinates that were distinct before transformation were rounded to the same coordinate after transformation.

Rolleroo, 2022-09-16:
Stuck the model I had issues with into Meshmixer, used the "Make Solid" command. Exported back out as an STL (it got quite a bit bigger after this) and Prusa seemed to cope with it after that. Hope this helps.

closed 2026-09-29 by prusaslicer-bot[bot]