prusa3d/PrusaSlicer #4064: Doesn't slice model stl file correct (cura does)

Asger-P, 2020-04-14:
Hi
I use PrusaSlicer Version 2.2  on Win 10 64 pro

You can see in the attached picture what is wrong Cura does it correct:
[image]

Here are the stl and 3mf file for the model
[Skruetrækker12x.zip](https://github.com/prusa3d/PrusaSlicer/files/4474777/Skruetraekker12x.zip)

Best regards
Asger

neophyl, 2020-04-14:
While Prusa Slicer doesn't report any errors with the model you can see there are even before its sliced as the mounting holes in the back are filled in.
If you repair the model through Netfabb it fixes it nicely and then when sliced it works to what I think it should be with 3 bevelled mounting holes and the 12 funnel shaped parts.

Asger-P, 2020-04-14:
Ups...
Sorry, I should have checked, I'l make sure I do that if there is a next time.

Thanks 
Best regards
Asger

closed 2020-04-14 by Asger-P