prusa3d/PrusaSlicer #4065: [FR]

qwewer0, 2020-04-14:
**Feature Request**

Is there a way to merge the first "move to next layer (0)" and the first "move to first skirt point" / "move to first perimeter point" or to switch the order, so the print head first goes to the desired XY position then to the Z position?

**Benefits:**
The print head won't leave a filament dot at the first "move to next layer (0)" position.

<details>
  <summary>Sample merged gcode segment</summary>
  
```
...
; End of custom gcode
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0 ; reset extrusion distance
G1 E-5.00000 F1800.00000 ; retract
G1 Z0.200 X50.000 Y50.000 F6000.000 ; move to first skirt point
G1 E5.00000 F1800.00000 ; unretract
...
```
  
</details>

rtyr, 2020-04-14:
Can you explain, why do you need it?

I don't think that switching the order is a good idea. Nozzle can be too low for safe travel on some printers (after homing). It is also very bad scenario for delta printers.

qwewer0, 2020-04-15:
> I don't think that switching the order is a good idea. Nozzle can be too low for safe travel on some printers (after homing). It is also very bad scenario for delta printers.

You are right, changing the order would cause a lot of trouble.

In my case, at the "move to next layer (0)" the nozzle would leave little filament dot, and if later in the printing job the nozzle crosses that point, then the nozzle picks it up and deposits it down somewhere else on the printed object, which isn't nice if it is on the surface.
But if the nozzle doesn't crosses that point then if I don't notice it at the end of the print and clean it up, that dot would stay there for the next print.

@rtyr Thanks for the response.

rtyr, 2021-02-12:
I believe we can close this. The issue seems easily avoidable by custom start g-code.

closed 2021-02-12 by rtyr

qwewer0, 2021-02-12:
> I believe we can close this. The issue seems easily avoidable by custom start g-code.

I'm ok with this closed, but it cannot be solved in the start gcode section as the problem comes after the start gcode.

rtyr, 2021-02-12:
Yes, but you can move XY wherever you want before it, richt?

qwewer0, 2021-02-12:
> Yes, but you can move XY wherever you want before it, richt?

If there is an easy way to automatically set an XY movement in start gcode to the print first XY position, then that would solve the problem by practically removing the first "to" print Z and XY movement, but I don't know if there is a way.

rtyr, 2021-02-12:
You can move the the corner of the bed for example, if only problem is small dot on the bed. Or just use the purge line. Maybe provide some example gcode file or 3MF project.

qwewer0, 2021-02-12:
> You can move the the corner of the bed for example, if only problem is small dot on the bed. Or just use the purge line. Maybe provide some example gcode file or 3MF project.

The problem was that the nozzle moves down to the first layer height then moves to the printing XY position and that movement can scraped the bed a bit, as my bed was pretty uneven. And it isn't a project specific problem, so I won't post 3mf file, if that is ok.

(I solved my bed problem, but I'm sure that I'm not the only one that has bad uneven bed surface.)

- If my memory serves me correctly, Cura solves this by having the "move to printing XYZ position" as one gcode line, which would be the same if we don't add a z lift at the end of the start gcode, but at leas we can do that.
- I PS, first there is a z movement which would erase any z lift in the start gcode, then moves to the printing XY position. 

But don't worry about it, this issue is closed and while others might have this problem, but at least now I don't.

rtyr, 2021-02-12:
I see, very uneven bed without mesh bed leveling. There was similar request in https://github.com/prusa3d/PrusaSlicer/issues/2931. I reopened that issue. Maybe we can do something about it.

qwewer0, 2021-02-12:
> I see, very uneven bed without mesh bed leveling. There was similar request in #2931. I reopened that issue. Maybe we can do something about it.

I will go and read through that.