prusa3d/PrusaSlicer #4066: Slicing Error in 2.2.0   (inside-outside mismatch)

hm17gh1, 2020-04-14:
### Version
Version: 2.2.0+linux-x64
Build: PrusaSlicer-2.2.0+linux-x64-202003211856
Operating System: Unix
System Architecture: 64 bit
System Version: Linux 4.15.0-96-generic x86_64
Total RAM size [MB]: 16,529MB
#
OpenGL installation
GL version: 3.0 Mesa 19.2.8
Vendor: Intel Open Source Technology Center
Renderer: Mesa DRI Intel(R) Iris Plus Graphics 640 (Kaby Lake GT3e) 
GLSL version: 1.30

### Operating system type + version
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.4 LTS
Release:	18.04
Codename:	bionic

### 3D printer brand / version + firmware version (if known)
2 variants of custom printer, but it is not relevant

### Behavior:
For the included model (openSCAD) the slicer mistakes the internal holes as object, and the object is not prezent in the output, as seen in the pictures.  This object works well in slic3r, cura, and older versions of prusaslicer.

* Steps needed to reproduce the problem:
Load the object, look at the preview

* Expected Results:
body of object and inside empty space should be correctly detected, not mismatched 

* Actual Results:
The object is not present, but inside empty spaces are sliced as if it is the object.
(screenshots, scad, stl, 3mf  attached)

[image]

[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs

[test-skew.zip](https://github.com/prusa3d/PrusaSlicer/files/4475795/test-skew.zip)

rtyr, 2020-04-14:
Model is not manifold. Repair it before slicing.

You can use Netfabb service to repair it (click on the icon):
[image]

Closing.

hm17gh1, 2020-04-14:
Model is produced by openSCAD and works well in EVERYTHING ELSE, including older versions of prusaslicer, cura, slic3r.  It is possibly wrongly detected as not-manifold due to float rounding error somewhere. Making prusaslicer dependent of external tools like netfab that does not work on linux is making me (and others) move away from it.

rtyr, 2020-04-14:
Well...PrusaSlicer, Netfabb, 3Dbuilder, Meshmixer...all reports your STL as broken...

Interesting thing here is that it actually slices fine even without repairing the model on my PC (win64), see screenshot.

OK, I will leave this open and label it for future improvements.

[image]

sboehle, 2020-04-20:
I can confirm the Issue using the files provided.
The *.3mf file is working - slicing is ok.
The *.stl file isn't working. Slicing is not ok.
When i use PrusaSlicer to repair the *.stl file, the fixed *.obj file looks like the "bad" sliced one.
Perhaps there is an issue with the autorepair/repair feature of PrusaSlicer?

Simplify3D can slice the "bad" stl without any issue. Cura, also.

FidelCapo, 2021-02-01:
This issue should be fixed in next release.
Closing.

closed 2021-02-01 by FidelCapo