prusa3d/PrusaSlicer #4067: Extruder turning off mid print

moboch, 2020-04-14:
### Version
_2.2.0_

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 with BigTreeTech Skr v.1.1 + TMC2208s in UART mode
Marlin 2.0.5.3
### Behavior
When i slice my files using Prusaslicer, the extruder just randomly shuts down at some point ( mostly around z = 10). Its not jamming and its also not overheating, it´s not even trying to push out the filament.I'm pretty sure it´s a slicer issue, since when i use Cura everything works flawlessly.
I´m using the built-in profiles for the Ender 3.

rtyr, 2020-04-14:
I don't think that this problem is slicer related. It will be most likely some hardware/firmware issue with your custom setup. I saw many reports like this from people using TMC2208s on the extruder (especially in the stealthchop mode). Other thing which comes to my mind is possible temperature drop triggering "prevent cold extrusion" feature. 

Anyway, please provide your 3MF project file.

moboch, 2020-04-14:
I can´t attach a 3mf directly so [here](https://workupload.com/file/euuAsnQQRvj) it is on Workupload. I also at first didn't believe it was a slicer issue, but it doesn't happen when using a different slicer like Cura or Ideamaker.
Neither the stepper itself nor the driver are overheating, they are both at around 30°C while printing.
I´ll have to check for a huge temperature drop in the nozzle, but i set the threshhold for that to 180°C and I´m usually printing at 215°C

rtyr, 2020-04-14:
I checked the gcode generated from your 3MF. It is OK, extrusion commands are there as expected. I even successfully printed that gcode on the Prusa MINI (with modified start gcode). So the issue is clearly not in the PrusaSlicer. 

There might be some differences, which are triggering that problem on your printer. For example, Cura is using absolute positioning for extruder by default. If I remember correctly, most of the issues with TMC2208 drivers on the extruder are not related to overheating, but rather to sudden movement changes of the extruder. I believe you will find some answers on the Marlin GitHub. 

Closing.

closed 2020-04-14 by rtyr