prusa3d/PrusaSlicer #4070: PrusaSlicer nozzle wipe during print (feature request)

bartlee01, 2020-04-14:
For prints that take up the entire Mk3S bed with lots of little parts, no matter how carefully the Z height is calibrated and how well the bed level compensation is set, there will sometimes be PETG stick to the outside of the nozzle and for tiny parts this can cause a problem when the blog eventually comes off. It can also leave burnt plastic stuck to parts or cause a nozzle crash and pause the print.

It would be great if PrusaSlicer had an option to use a corner of the print bed to print a little tower that wipes the nozzle each time the z height moves up an increment. It could be a little square where after printing the perimeter the nozzle moves out a little and down and then moves around the perimeter again to wipe off any ooze that is stuck to the outside of the nozzle. It could do this for every layer increment to stop a large blog of PETG ever developing.

neophyl, 2020-04-15:
Couldnt you just do this by adding gcode to either after layer or before layer section under Printer settings ?  Any gcode added in there gets inserted into the output each time the layer changes.

bartlee01, 2020-04-15:
> Couldnt you just do this by adding gcode to either after layer or before layer section under Printer settings ? Any gcode added in there gets inserted into the output each time the layer changes.

Maybe this is possible. I have never written custom gcode though. If it is possible then I think it would be quite a useful feature. I've even waited until a few layers are printed, paused and cleaned the nozzle, resumed the print and with some PETG the nozzle still slowly gets plastic on the outside again and it eventually comes off, sometimes in a way that damages the print or makes the nozzle crash.

This is happening with Prusa PETG (the worst is the white plastic). Other ones such as Prusament Neon Green don't build up on the nozzle at all so it's not the flow rate and to turn it low enough not to build up it is definitely under extruding which makes the print weaker (no good for mechanical parts I print). I have tried changing the print temperature between 230 up to 250 degrees but it still happens, especially for white.

espr14, 2020-04-15:
Will wiping help? If so, you can enable wipe during retraction and prolong retraction.

I think wipe won't help since the extra filament is stuck on the side of the nozzle. You would have to set much higher temperature to make filament slide down or attach a brush at the end of an axis which will do the job automatically.

johnbiggs, 2020-04-23:
Drop your first layer acceleration and it helps with petg a lot.

espr14, 2020-04-23:
First layer is not the only issue. It happens to me as well in higher layers where infills of ~1 mm^2 are done. These cause overextrusion / oozing so the reminding material sticks to nozzle. Standard wiping does not take whole object into account so it wipes at the place which is not enough.

mepoland, 2022-12-16:
I print a lot of small parts with PETG and have the same issue. This feature would be a major time saver for me; currently I have to make sure I can go in and manually pause/clean/resume every 8ish hours or else there is sure to be a blob big enough to knock something loose and ruin the print. I think this would need to be coded into the slicer itself rather than a simple custom G code; the wiping tower @bartlee01 describes, which I think would work well, requires some funky Z-direction moves.

closed 2026-09-29 by prusaslicer-bot[bot]