prusa3d/PrusaSlicer #4071: secondary hard drive now hunts

DOUG888, 2020-04-14:
### Version
_2-2-0_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Windows7_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_Prusa Mk3s_

### Behavior
* _Second hard drive is now hunting_
* _Installed latest version of Prusa slicer and drivers_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

JohnnyDeer, 2020-04-15:
Hi, can you please describe more about issue and steps needed to reproduce it?

DOUG888, 2020-04-15:
Once version 2.2.0 of the prusaslicer it was installed immediately my second hard drive began to hunt I removed in sequence the drivers package which made no difference immediately that I removed the prusa slicer latest edition version the drive hunting stopped I reverted back to the previous version and there was no issues. There appears to be some incompatibility with the latest version of pressure slicer and having a second drive installed

neophyl, 2020-04-15:
Possibly the same issue that caused the floppy drive access.  #3967
https://forum.prusaprinters.org/forum/prusaslicer/prusaslicer-trying-to-access-my-floppy-disk-a/  

Bubnikv posted on that thread and explained what was happening.

DOUG888, 2020-04-15:
Yes there is a good chance my problem is related to all the linkage that you supplied I tend to agree that the software that prusa developed is excellent and that the response time and services that are supplied are also excellent but in this case I think maybe just pressing a button would simplify the process and remove this potential bug so that you can then manually load or download from the required device without constantly polling all the drives available or listed such as secondary hard drives or floppy drives thanks again for all the support.

bubnikv, 2020-10-09:
PrusaSlicer newly registers for removable media insert / eject notifications, therefore the devices are no more scanned regularly. Thanks @mjgtp from prusaprinters.org, see [the following discussion](https://forum.prusaprinters.org/forum/prusaslicer/prusaslicer-trying-to-access-my-floppy-disk-a)

It should be fixed in the upcoming PrusaSlicer 2.3.0-alpha1.
Closing.

closed 2020-10-09 by bubnikv