prusa3d/PrusaSlicer #4072: Layer heaight and count Preview Issue

MichaelMD78, 2020-04-15:
### Version 2.2.0+win64
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version Windows 10 latest release
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known) Custom Build RepRap
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
First layer height set to 0.2 Layer height set to 0.32, however the Preview shows first layer at 0.2 and the second layer at 0.44 (it should be 0.52) Preview shows layer count at 88, but the gcode file shows the layer count at 73
* _Steps needed to reproduce the problem_
Explained above
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

[image]

[image]

[image]

_Is this a new feature request?_No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[Skimmer Bucket v14_Clamp.zip](https://github.com/prusa3d/PrusaSlicer/files/4479788/Skimmer.Bucket.v14_Clamp.zip)

rtyr, 2020-04-15:
It is actually correct, because PrusaSlicer can use independent (different) layer height for the supports. As you can see in the preview, there are supports printed at 0.44. The layer count in the preview is a sum of the object layers and support layers.

MichaelMD78, 2020-04-15:
Ok thank you, I see that when the supports are turned off then the counts
are all correct. Thank you for the clarification.

rtyr, 2020-04-15:
You can also set layer height range for the supports here (you might want to increase max value a little bit):

[image]

Closing.

closed 2020-04-15 by rtyr