prusa3d/PrusaSlicer #4073: Specify minimum length for gap infill

smurfix, 2020-04-15:
### Version
2.2, current

### Behavior
While some gap infills are relevant, others are stupid. I need a gap fill that glues two edges together; I do not need a gap infill that is 0.2mm long, helps nobody, and creates a blob that breaks my print when I do the next layer.

See e.g. this screenshot: the gap fill on the bottom left makes sense, the others do not.

[image]

Since it's not always possible to avoid (some) gaps entirely, a print settings option that requires a minimum length for gap fills would be a good way to fix the problem.

_Is this a new feature request?_ yes

neophyl, 2020-04-15:
Well this is the fourth issue I've found for this.  Seems to be popular lol.  Someone over on the Prusa forum was just asking the same thing too.
2780
2691
1806

paulnewall, 2020-04-18:
I would find this feature very helpful for PETG. The short gap fills create bumps that interfere with the next layer.

LeavOA, 2020-08-29:
I'll toss my vote in as well. causes issues with ABS on the volacno (ooze festival...)

neophyl, 2020-08-29:
You all could just use Super Slicer (its a fork of Prusa Slicer available https://github.com/supermerill/SuperSlicer/releases ) as that has this in already.  While some of the settings are arranged slightly differently they are all there (and loads more).  The one you want is on Print Settings>Perimeters & Shell>Gap Fill > Min surface.  Settable as a size in mm or a % of nozzle width.

I think the best description I've seen of it is - Prusa Slicer on steroids.

nietaki, 2021-03-20:
When printing PETG things like this are an unnecessary liability, sometimes choosing between disabling gap fill and leaving it on is difficult :)

[image]

bringbackgroupb, 2021-04-03:
I've got this issue as well. Doesn't seem like it would be too hard to implement, the slicer is already able to distinguish the features which are gap fill, we just need an option to omit certain ones.

[image]

CamStLouis, 2022-08-19:
100% Agree. I've got a huge tall cylinder I can't resize for acoustic reasons, and having gap fill enabled would be useless. However, disabling it then makes the walls where it joins the mouthpiece just one extrusion thick :( 
[image]

stepikovo, 2022-08-29:
What about using the new 2.5.0 slicer with the Arachne engine? That could help.

CamStLouis, 2022-09-03:
> What about using the new 2.5.0 slicer with the Arachne engine? That could help.

From the results in the development thread, that seems nowhere near ready for primetime. Again, just the option to specify what size of gap is worth filling would solve his whole issue.

DragonFinlay, 2022-09-09:
> From the results in the development thread, that seems nowhere near ready for primetime. Again, just the option to specify what size of gap is worth filling would solve his whole issue. I think it's a pretty reasonable feature request.

I agree that choosing a gap size threshold would be useful, but as we're switching to Arachne, new features for the classic perimeter generator will get a lower priority.

Also, no offense, but why don't you give it a try and see for yourself if it's suitable for you or not? :)
One thing is for sure, Arachne _will_ get rid of your gap fill issues.

[image]

CamStLouis, 2022-09-09:
Just tried a print; Arachne has already fucked up my seam placement (I selected “rear” but got diagonal) and seems to have poorer dimensional stability. The ring of ripples on the part from overzealous gap fill is gone, but features that formerly came out crisp are a little rougher this time around. I use this for work so unfortunately the aesthetics matter.

thenickdude, 2022-09-09:
I print a thin-walled tube with Arachne and end up with two perimeters that don't touch each other when printed, and the whole thing crumbles when squeezed. Good exterior surface finish though compared to the lumpiness I got with gap fill.

DragonFinlay, 2022-09-15:
> Just tried a print; Arachne has already fucked up my seam placement (I selected “rear” but got diagonal) and seems to have poorer dimensional stability. The ring of ripples on the part from [...]

Right, I understand. We are currently working on seam placement when using Arachne, so this should be fixed soon. Please refer to #8443, #8512, #8686 and #8841 (we will close some of these soon to keep discussion to one thread).

> I print a thin-walled tube with Arachne and end up with two perimeters that don't touch each other when printed, and the whole thing crumbles when squeezed. Good exterior surface finish though compared to the lumpiness I got with gap fill.

This is also something we're aware of, please check #8578.

trackhacs, 2023-02-10:
> One thing is for sure, Arachne will get rid of your gap fill issues.

For me it actually introduces gap fill issues ... which are not here with classic implementation.

jwidess, 2023-12-15:
Ran into an issue today I've had many times before, **gap fill creating tiny extrusions** that do nothing but cause problems. I was originally using the **Arachne** perimeter generation, but that was **ALSO** causing very small extrusions at the tips of perimeters, so I swapped to Classic and instead I'm now getting these tiny gap fills, ugh. Really hope this issue can be resolved soon as this has been a problem for years and, as mentioned in [this comment](https://github.com/prusa3d/PrusaSlicer/issues/4073#issuecomment-683316776), SuperSlicr already has a solution for this.
| Classic (White Gap Fills)  | Arachne (Tiny Yellow Extrusions) |
|--------|--------|
| [image]| [image] |

closed 2026-09-29 by prusaslicer-bot[bot]