prusa3d/PrusaSlicer #4074: Feature Request: Can support enforcers add supports even when it is turned off in Print Settings?

kellybyrd, 2020-04-15:
### Version
2.2.0+ 

### Operating system type + version
MacOS 10.15.4

### 3D printer brand / version + firmware version (if known)
Prusa I3 MK3S - OEM firmware v3.8.1

### Behavior
**This a new feature request?**
Overall: I would like to keep the _Generate Support Material_ option turned off and then use _Support Enforcer_ objects to add supports to areas I think need them.

Details:
I have several _Print Settings_ profiles, including the ones built in for my MK3S printer. What I would like to do is keep the _Generate Support Material_ option off on all of these (which is the default), but then for the few times I need supports, add them to specific areas using _Support Enforcer_ objects. Right now, if _Generate Support Material_ is off, then _Support Enforcers_ don't appear to do anything.

rtyr, 2020-04-15:
You can use this setting for that purpose. Can we close this issue?

[image]

kellybyrd, 2020-04-15:
🤦 Ouch. I cannot believe I didn't check that. 
Thanks!

closed 2020-04-15 by kellybyrd