prusa3d/PrusaSlicer #4080: Minimum shell thickness appears to be rounding up when evenly divisible

prenetic, 2020-04-16:
### Version
2.2.0+win64

### Operating system type + version
Windows 10 version 1909

### 3D printer brand / version + firmware version (if known)
Wanhao i3 Plus clone (ADVi3pp 4.0.5 firmware, Marlin-based)

### Behavior
In some cases, the computed "Minimum shell thickness" values, when evenly divisible by layer height, appear to be rounding up to the next evenly divisible number.

[image]

A 2.16 mm horizontal shell thickness should be attainable with 6 layers at a 0.36 mm layer height, however the GUI indicates this is being rounded up to 2.52 (an additional layer is added).

[image]

When specifying a desired minimum shell thickness of 2.1599999 mm, this is rounded up to 2.16 mm as expected, which further indicates 2.16 should be a valid minimum shell thickness without rounding up to 2.52.

#### Project File (.3MF) where problem occurs
[buddha_stone.zip](https://github.com/prusa3d/PrusaSlicer/files/4489364/buddha_stone.zip)

johnbiggs, 2020-04-23:
Something is wrong there but you seem to forget the first layer is 0.48mm. So the bottom shell would have to round to 2.28mm.

prenetic, 2020-04-29:
Good point, I missed that entirely. As you say though, the 2.52 mm value seems off.

closed 2026-09-29 by prusaslicer-bot[bot]